Hexxeh/rpi-firmware #71: Kernel PPS Support Missing

darkomenz, 2015-01-21:
commit cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45

http://www.satsignal.eu/ntp/Raspberry-Pi-NTP.html

popcornmix, 2015-01-21:
Added this: https://github.com/raspberrypi/linux/commit/a96c657cb56bc612f4dec5ca2faf0bda19e876cc
Will be in next build.

popcornmix, 2015-01-22:
Firmware is updated. Please test after running rpi-update.

darkomenz, 2015-01-22:
root@NTP-0:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  7811  100  7811    0     0  28965      0 --:--:-- --:--:-- --:--:-- 40056
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    437      0 --:--:-- --:--:-- --:--:--   589
100 25.4M  100 25.4M    0     0   478k      0  0:00:54  0:00:54 --:--:-- 99776
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 3.18.3+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to ebc44216ee5e3877828c0c6eac6a1341202e768e
 *** A reboot is needed to activate the new firmware

[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xe bcm2708.serial=0xa174d039 smsc95xx.macaddr=B8:27:EB:74:D0:39 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait smsc95xx.turbo_mode=0 bcm2708.pps_gpio_pin=18
[   10.513457] pps_core: LinuxPPS API ver. 1 registered
[   10.517504] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[   25.990638] pps_ldisc: PPS line discipline registered
[   25.994832] pps pps0: new PPS source ttyAMA0
[   25.995130] pps pps0: source "/dev/ttyAMA0" added

root@NTP-0:~# lsmod
Module                  Size  Used by
pps_ldisc               2285  3
pps_gpio                2897  0
pps_core                8752  4 pps_ldisc,pps_gpio
snd_bcm2835            21342  0
snd_pcm                93100  1 snd_bcm2835
snd_seq                61097  0
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    67211  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
uio_pdrv_genirq         3666  0
uio                     9897  1 uio_pdrv_genirq

The gpio based pps device is missing. This issue is still not resolved.

PeterPablo, 2015-01-22:
@darkomenz, please be aware of the Device Tree introduced by Kernel 3.18.*. You need to manually enable GPIO for pin 24 with a [new method](http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314).

darkomenz, 2015-01-22:
Hello,

None of the existing overlays enable the pps-gpio driver. I would write one if I was familiar with device tree but unfortunately its a new concept to me. That being said could someone that is familiar with the concept write one and add it to the supported list of overlays.

https://www.kernel.org/doc/Documentation/devicetree/bindings/pps/pps-gpio.txt
https://www.kernel.org/doc/Documentation/devicetree/bindings/gpio/gpio.txt

Thanks

darkomenz, 2015-01-22:
From reading information about the device tree system it seems that the overlay may look something like this.
However my example is incomplete and may be incorrect as its just a guess.

/dts-v1/;
/plugin/;
/ {
    compatible = "brcm,bcm2708";
    fragment@0 {
        __overlay__ {
            pps@0 {
                compatible = "pps-gpio";
                status = "okay";
            };
        };
    };
    fragment@1 {
        target = <0xdeadbeef>;
        pps_gpio_pin {
            brcm,pins = <0x18>;
            brcm,function = <0x0>;
            brcm,pull = <0x0>;
        };
    };
    __overrides__ {
        gpios = <&pps_gpio_pin>,"brcm,pins:24";
    };
};

popcornmix, 2015-01-23:
ping @pelwell

pelwell, 2015-01-23:
Hey, that was a pretty good attempt. All fragments need valid targets and __overlays__, and the number after the colon in the __overrides__ section is a byte offset - otherwise it's OK. The 0xdeadbeef is essentially an unresolved symbol that gets patched by the loader. In the source it should be a reference to a label.

Here's my tidied up version:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";
	fragment@0 {
		target-path = "/";
		__overlay__ {
			pps: pps {
				compatible = "pps-gpio";
				pinctrl-names = "default";
				pinctrl-0 = <&pps_pins>;
				gpios = <&gpio 24 0>;
				status = "okay";
			};
		};
	};

	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			pps_pins: pps_pins {
				brcm,pins =     <24>;
				brcm,function = <0>;    // in
				brcm,pull =     <0>;    // off
			};
		};
	};

	__overrides__ {
		gpios = <&pps>,"gpios:4",
		      	<&pps_pins>,"brcm,pins:0";
	};
};
```

The name change (pps_gpio_pin -> pps_pins) is just for consistency with the rest of the .dtb and overlays.

The overlay compiles and loads and appears to do the right thing, but I'm not in a position to test it. This is the result:
```
0A3+7QAAA98AAAA4AAADVAAAACgAAAARAAAAEAAAAAAAAACLAAADHAAAAAAAAAAAAAAAAAAAAAAA
AAABAAAAAAAAAAMAAAANAAAAAGJyY20sYmNtMjcwOAAAAAAAAAABZnJhZ21lbnRAMAAAAAAAAwAA
AAIAAAALLwAAAAAAAAFfX292ZXJsYXlfXwAAAAABcHBzAAAAAAMAAAAJAAAAAHBwcy1ncGlvAAAA
AAAAAAMAAAAIAAAAF2RlZmF1bHQAAAAAAwAAAAQAAAAlAAAAAQAAAAMAAAAMAAAAL96tvu8AAAAY
AAAAAAAAAAMAAAAFAAAANW9rYXkAAAAAAAAAAwAAAAQAAAA8AAAAAgAAAAMAAAAEAAAAQgAAAAIA
AAACAAAAAgAAAAIAAAABZnJhZ21lbnRAMQAAAAAAAwAAAAQAAABK3q2+7wAAAAFfX292ZXJsYXlf
XwAAAAABcHBzX3BpbnMAAAAAAAAAAwAAAAQAAABRAAAAGAAAAAMAAAAEAAAAWwAAAAAAAAADAAAA
BAAAAGkAAAAAAAAAAwAAAAQAAAA8AAAAAQAAAAMAAAAEAAAAQgAAAAEAAAACAAAAAgAAAAIAAAAB
X19vdmVycmlkZXNfXwAAAAAAAAMAAAAcAAAALwAAAAJncGlvczo0AAAAAAFicmNtLHBpbnM6MAAA
AAACAAAAAV9fc3ltYm9sc19fAAAAAAMAAAAcAAAAcy9mcmFnbWVudEAwL19fb3ZlcmxheV9fL3Bw
cwAAAAADAAAAIQAAAHcvZnJhZ21lbnRAMS9fX292ZXJsYXlfXy9wcHNfcGlucwAAAAAAAAACAAAA
AV9fZml4dXBzX18AAAAAAAMAAAA5AAAAgC9mcmFnbWVudEAwL19fb3ZlcmxheV9fL3BwczpncGlv
czowAC9mcmFnbWVudEAxOnRhcmdldDowAAAAAAAAAAIAAAABX19sb2NhbF9maXh1cHNfXwAAAAAA
AAADAAAAVwAAAIUvZnJhZ21lbnRAMC9fX292ZXJsYXlfXy9wcHM6cGluY3RybC0wOjAAL19fb3Zl
cnJpZGVzX186Z3Bpb3M6MAAvX19vdmVycmlkZXNfXzpncGlvczoxMgAAAAAAAgAAAAIAAAAJY29t
cGF0aWJsZQB0YXJnZXQtcGF0aABwaW5jdHJsLW5hbWVzAHBpbmN0cmwtMABncGlvcwBzdGF0dXMA
bGludXgscGhhbmRsZQB0YXJnZXQAYnJjbSxwaW5zAGJyY20sZnVuY3Rpb24AYnJjbSxwdWxsAHBw
cwBwcHNfcGlucwBncGlvAGZpeHVwAA==
```

Put it in a file (e.g. pps-gpio-overlay.b64) and run:
```bash
base64 -d pps-gpio-overlay.b64 > pps-gpio-overlay.dtb
```
Then copy pps-gpio-overlay.dtb into /boot/overlays, and add the following to your config.txt:
```
dtoverlay=pps-gpio,gpios=24
```
The pin assignment to 24 is optional - it is the default in the overlay.

Let us know how you get on - if it works we'll include it in the firmware.

darkomenz, 2015-01-23:
Kernel without device tree:

[    9.928027] pps_core: LinuxPPS API ver. 1 registered
[    9.929536] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    9.943891] pps pps0: new PPS source pps-gpio.18
[    9.945398] pps pps0: Registered IRQ 188 as PPS source

ppstest works fine.

Kernel with device tree and dtoverlay=pps-gpio,gpios=24:

[    4.726178] pps_core: LinuxPPS API ver. 1 registered
[    4.754534] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    4.837209] pps pps0: new PPS source pps.-1
[    4.839491] pps pps0: Registered IRQ 418 as PPS source

root@NTP-0:~# ppstest /dev/pps0
trying PPS source "/dev/pps0"
found PPS source "/dev/pps0"
ok, found 1 source(s), now start fetching data...
time_pps_fetch() error -1 (Connection timed out)

pelwell, 2015-01-25:
@darkomenz Can you confirm which GPIO pin the PPS device is attached to? The NTPI board uses pin 18 and the command line parameter agrees, but @PeterPablo 's comment and your proposed overlay used 24 (which, perhaps uncoincidentally, is "18" in hex). Alternatively, perhaps there is some confusion between GPIO numbering and header pin numbering; GPIO 18 is found on header pin 12, and GPIO 24 is found on header pin 18, but the APIs deal in GPIO numbers.

Unless you are certain that GPIO 24 is correct, try changing the dtoverlay line to read:
```
dtoverlay=pps-gpio,gpios=18
```

darkomenz, 2015-01-25:
Kernel with device tree and dtoverlay=pps-gpio,gpios=18:

[    4.708624] pps_core: LinuxPPS API ver. 1 registered
[    4.746069] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    4.810198] pps pps0: new PPS source pps.-1
[    4.828161] pps pps0: Registered IRQ 412 as PPS source

root@NTP-0:~# ppstest /dev/pps0
trying PPS source "/dev/pps0"
found PPS source "/dev/pps0"
ok, found 1 source(s), now start fetching data...
source 0 - assert 1422203534.999794014, sequence: 100 - clear  0.000000000, sequence: 0
source 0 - assert 1422203535.999791894, sequence: 101 - clear  0.000000000, sequence: 0
source 0 - assert 1422203536.999790775, sequence: 102 - clear  0.000000000, sequence: 0
source 0 - assert 1422203537.999787656, sequence: 103 - clear  0.000000000, sequence: 0

Yay!

 @PeterPablo 's comment confused me but I ran with it was not sure if it was supposed to be in hex or not. When this gets added as a overlay please make pin 18 the default since its referenced everywhere on the internet.

pelwell, 2015-01-25:
Nice. It will probably get released tomorrow, with 18 as the default.

pelwell, 2015-01-26:
This is in the firmware repo now. Note that I renamed the parameter from "gpios" to "gpiopin", to match  other overlays.

Ruffio, 2015-06-21:
@pelwell @darkomenz so this can be closed now?

pelwell, 2015-06-22:
Close it. The issue has been dealt with.

closed 2015-06-22 by popcornmix