Hexxeh/rpi-firmware #72: issues with smbus

bootsmann, 2015-01-23:
After updating with ``rpi-update`` to "Linux raspberrypi 3.18.3+ #740 PREEMPT Wed Jan 21 23:55:56 GMT 2015 armv6l GNU/Linux" I have some issues with smbus python mudule. I am using pi-pan kit an pilight from http://www.openelectrons.com/index.php?module=pagemaster&PAGE_user_op=view_page&PAGE_id=24 Openelectrons

    Traceback (most recent call last):
      File "check-pilight.py", line 32, in <module>
        pl = pilight.PILIGHT()
      File "/root/pi-pan/pilight.py", line 61, in __init__
        OpenElectrons_i2c.__init__(self, pilight_address >> 1)
      File "/root/pi-pan/OpenElectrons_i2c.py", line 54, in __init__
        self.bus = smbus.SMBus(b)
    IOError: [Errno 2] No such file or directory

Srokap, 2015-02-04:
Same here, i2c-dev doesn't mount any devices in `/dev/i2c-*`

pelwell, 2015-02-04:
Add:
```
dtparam=i2c=on
```
to /boot/config.txt, and reboot.

Or use raspi-config to do the same (Advanced Options, I2C, ...)

See here for an explanation: http://www.raspberrypi.org/forums/viewtopic.php?f=44&t=98318

pelwell, 2015-02-09:
Can we close this issue?

bootsmann, 2015-02-09:
The issue is over (for me)
it runs with 
    dtparam=i2c=on

closed 2015-02-09 by popcornmix

neuberfran, 2018-01-16:
Please help me

[image]

pelwell, 2018-01-16:
There isn't much to go on there:

1. Which Pi model are you using?

2. What does `uname -a` return?

3. Which I2C bus is your device on?

4. What is its I2C address?

5. Has this ever worked before? If so, what has changed?

To save time I'm going to guess that your device is 0x1b on i2c1, which is equivalent to 0x36 on the wire. The SmartDemo library I found uses 0x36/0x1b as the default address, and it should correctly work out to use i2c1 on your Pi unless you are using a very old Model A or B.

I suggest you either add some debugging print statements to the scripts or run the command under pdb, the python debugger.

neuberfran, 2018-01-16:
I2C bus is one I think

I stay using rpi 3

I android things I detected i2c device but motors don't run more today. Yesterday the motors run with this command: **pio i2c I2C1 0x1B write-reg-byte 0x41 0x53**

[image]

pelwell, 2018-01-16:
Use the debugger - run `pdb SmartDrive-demo.py` then type `?` to get a list of commands, but the basics are:
* `n` - run to the next line
* `s` - like `n`, but steps into a function/method call
* `c` - continue, allow the script to run
* `p` - can be used to display variables
* `l` - list the program, normally around the current execution point
etc.

neuberfran, 2018-01-16:
[image]

neuberfran, 2018-01-16:
@pelwell  Can you help me with this error?

 File "/usr/local/lib/python2.7/dist-packages/OpenElectrons_i2c.py", line 61, in __init__
    **_self.bus = smbus.SMBus(b)_**
IOError: [Errno 2] No such file or directory

pelwell, 2018-01-16:
What is the value of b when it fails?

neuberfran, 2018-01-16:
@pelwell Honestly I do not know. Could you help figure it out?

(according to this site: https://pypi.python.org/pypi/smbus-cffi/0.4.1
I try pip install smbus-cffi  and pip install cffi **but not solved**

pelwell, 2018-01-16:
Follow these simple steps:
```
$ pdb SmartDrive-demo.py
...
(Pdb) c
...
(Pdb) p path
(Pdb) p b
```
One of the `p` commands might fail, but I'm hoping that one will return an answer.

Alternatively, try:
```
$ sudo apt-get install strace pastebinit
$ strace SmartDrive-demo.py | pastebinit
```
Then post the URL it displays, which will be something like `http://paste.debian.net/1005554`.

neuberfran, 2018-01-16:
[image]

neuberfran, 2018-01-16:
[image]

neuberfran, 2018-01-16:
I solved when I change value of the self.bus = **smbus.SMBus(b)** to **self.bus = smbus.SMBus(1)**

now I have a problem do run python explorer1.py because I can't install explorerlib using pip

[image]

pelwell, 2018-01-16:
Where did you get explorer.py and explorer1.py?

neuberfran, 2018-01-17:
[image]
[image]
[image]

TkS
I get explorer1.py and explorer.py from OpenElectrons.com website has merged with mindsensors.com.

explorer1.py

https://pastebin.com/TTfBCF92

explorer.py

https://pastebin.com/zTV3aAN7

pelwell, 2018-01-17:
Your cffi installation is failing because you don't have a gcc toolchain - run `sudo apt-get install gcc` and try the installations again.

neuberfran, 2018-01-18:
[image]
[image]

pelwell, 2018-01-18:
`sudo apt-get install libffi-dev`

neuberfran, 2018-01-18:
after  sudo apt-get install libffi-dev

[image]

[image]

pelwell, 2018-01-18:
Ho hum. There is a package called python-cffi that you could apt-get install. But then you'll have to find somebody else to help you - I'm not much of a Python expert, I don't know the OpenElectrons platform, and this isn't the right place for this discussion.

neuberfran, 2018-01-18:
https://github.com/androidthings/contrib-drivers/issues/70

neuberfran, 2018-01-18:
i need freelancer