Hexxeh/rpi-firmware #74: Latest updates kill Sonic Pi: it freezes

rbnpi, 2015-02-06:
Sonic Pi works fine on a fresh Raspian release 31/1/2015 on rpi2. I did an RPi-update this afternoon and it now freezes whe you start playing. Sam Aaron is aware but doesn't know why.

AndrewFromMelbourne, 2015-02-07:
This is discussed on the Raspberry Pi forum (http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=98367&start=175#p685622).

rbnpi, 2015-02-07:
Thanks for the above. Which firmware ID should I downgrade to to restore Sonic-Pi use?

AndrewFromMelbourne, 2015-02-07:
You only have four to choose from as support for RPI2 was added on 3 Feb. So

c157458351071104eaa8afa7313f2f26afb499da - Add support for Raspberry Pi 2
2d2052a9905a91a90f565fa9925e7ce987a6b423 - Add gpu_mem_1024 config option
a8b36a5848f1c236fc7d9f48d8fc1ea758fa5796 - kernel: Add USBIP modules
93cb9a45741681987b207c68b63db7ab72b2a4cf - bcm2709: Set the "always-on" timer flag ...

I would try the second one at a guess. I can't try it myself, my RPI2 hasn't arrived yet :(

pelwell, 2015-02-07:
The problem is triggereded by the 2/2 User/Kernel VM split, which is necessary (at least for now) in order for the ARMs to be able to address more than 750MB. If you revert that change (see below) then you will limit the ARMs to 750MB (although if you set GPU mem to 250MB then you don't actually lose anything).

SonicPi does work on the kernel shipping with the current Raspbian release, because it has the old 3/1 split, so you could install that instead. Alternatively, use menuconfig to reselect 3G before rebuilding your kernel yourself.

popcornmix, 2015-02-27:
The Sonic Pi issue was a bug in Sonic Pi (an uninitialised variable) and is fixed in the latest version.
It was only coincidental that the updated firmware/kernel exposed the issue.

closed 2015-02-27 by popcornmix