Hexxeh/rpi-firmware #75: fsck fat cannot allocate memory

rickysarraf, 2015-02-09:
On my FAT file systems, I keep getting the following memory allocation error, when trying to run fsck. The ext4 fsck tool works perfect on the other drive.

Linux raspberrypi 3.18.6+ #1 PREEMPT Sat Feb 7 18:11:00 GMT 2015 armv6l GNU/Linux
pi@raspberrypi ~ $ sudo fsck -t fat /dev/sdb1
fsck from util-linux 2.20.1
dosfsck 3.0.13, 30 Jun 2012, FAT32, LFN
malloc:Cannot allocate memory

Since this is so core to the kernel, I couldn't think of anywhere else to file a bug report.

popcornmix, 2015-02-09:
What does:
```
vcgencmd get_mem arm 
```
report?

rickysarraf, 2015-02-09:
pi@raspberrypi ~ $ vcgencmd get_mem arm
arm=256M

I just read up now on this article from 2002. http://archive.linuxgizmos.com/why-is-malloc-different-under-uclinux/
Does it apply to RPi too ?

popcornmix, 2015-02-09:
Are you using uclinux? (default raspbian doesn't).
Can you change in config.txt:
```
gpu_mem=16
```
and reboot and try?

I think the issue is that fsck requires RAM proportional to disk size, and if the disk is big and RAM is small it will fail. E.g.
http://www.linuxquestions.org/questions/linux-software-2/dosfsck-does-not-work-on-1-5-tb-hard-disk-any-alternative-836039/

rickysarraf, 2015-02-09:
It still fails. :-(

I do have a large drive, 2 TB.

pi@raspberrypi ~ $ sudo fsck.vfat -a /dev/sdb1
dosfsck 3.0.13, 30 Jun 2012, FAT32, LFN
malloc:Cannot allocate memory
pi@raspberrypi ~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           371        215        156          0         25         86
-/+ buffers/cache:        103        268
Swap:           99          0         99
pi@raspberrypi ~ $ sudo fdisk -l /dev/sdb

Disk /dev/sdb: 2000.4 GB, 2000398931968 bytes
255 heads, 63 sectors/track, 243201 cylinders, total 3907029164 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x42b7cbd5

   Device Boot      Start         End      Blocks   Id  System
/dev/sdb1            2048  3460302847  1730150400    c  W95 FAT32 (LBA)
/dev/sdb2      3460302848  3907029163   223363158   83  Linux
pi@raspberrypi ~ $ vcgencmd get_mem arm
arm=384M

rickysarraf, 2015-02-09:
In case if you need the config.txt too.

i@raspberrypi ~ $ cat /boot/config.txt
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16  
#overscan_right=16 
#overscan_top=16   
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=900

# License keys
decode_MPG2=0x750ae72a
decode_WVC1=0x4455315c

# for more options see http://elinux.org/RPi_config.txt

# NOOBS Auto-generated Settings:
hdmi_force_hotplug=1
config_hdmi_boost=4
overscan_left=24   
overscan_right=24  
overscan_top=16
overscan_bottom=16 
disable_overscan=0 
gpu_mem=16
core_freq=250
sdram_freq=450
over_voltage=2
start_x=1

popcornmix, 2015-02-09:
Remove start_x=1 - that stops gpu_mem=16 from working.

rickysarraf, 2015-02-09:
It still fails. But now, it takes a while before failing on malloc. Earlier malloc() used to fail instantly. I guess given the large drive, I'm OOM ??? And I doubt if adding SWAP will help ?

pi@raspberrypi ~ $ time sudo fsck -a -vvv /dev/sdb1
fsck from util-linux 2.20.1
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13, 30 Jun 2012, FAT32, LFN
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "mkdosfs"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
     16384 bytes per cluster
        32 reserved sectors
First FAT starts at byte 16384 (sector 32)
         2 FATs, 32 bit entries
 432340992 bytes per FAT (= 844416 sectors)
Root directory start at cluster 2 (arbitrary size)
Data area starts at byte 864698368 (sector 1688864)
 108081623 data clusters (1770809311232 bytes)
63 sectors/track, 255 heads
         0 hidden sectors
3460300800 sectors total
malloc:Cannot allocate memory

real    0m16.202s
user    0m0.040s
sys     0m5.250s

By the way, by disabling X, I was able to get the actual memory listing.

pi@raspberrypi ~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           482         41        441          0          1         14
-/+ buffers/cache:         25        457
Swap:           99          0         99

popcornmix, 2015-02-09:
swap would probably help. You could also try e2fsck
```
sudo apt-get install e2fsck-static
```

rickysarraf, 2015-02-09:
Adding 500 MiB of swap prolonged the malloc failure, but still happens. Next I'll try with the static binary.

pi@raspberrypi /var $ time sudo fsck -a -vvv /dev/sdb1
fsck from util-linux 2.20.1
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13 (30 Jun 2012)
dosfsck 3.0.13, 30 Jun 2012, FAT32, LFN
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "mkdosfs"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
     16384 bytes per cluster
        32 reserved sectors
First FAT starts at byte 16384 (sector 32)
         2 FATs, 32 bit entries
 432340992 bytes per FAT (= 844416 sectors)
Root directory start at cluster 2 (arbitrary size)
Data area starts at byte 864698368 (sector 1688864)
 108081623 data clusters (1770809311232 bytes)
63 sectors/track, 255 heads
         0 hidden sectors
3460300800 sectors total
/Video/MyVid-212134.mp4
  File size is 325386240 bytes, cluster chain length is > 325386240 bytes.
  Truncating file to 325386240 bytes.
Reclaiming unconnected clusters.
malloc:Cannot allocate memory

real    12m49.803s
user    1m26.440s
sys     1m0.360s
pi@raspberrypi /var $ free -m
             total       used       free     shared    buffers     cached
Mem:           482         28        454          0          1          8
-/+ buffers/cache:         18        464
Swap:          499          7        492

rickysarraf, 2015-02-09:
Actually, there's no static fsck for FAT. e2fsck is only for ext file systems.

BTW, do you want to keep this bug report open ? I don't know if this can reach a terminal state.

popcornmix, 2015-02-09:
Pretty sure there is no fix from the firmware or kernel side.
It's an issue with the fsck app requiring memory proportional to disk size.
Lots of hits if you search for:
```
fsck "malloc:Cannot allocate memory"
```
It's just trying to find if any of those reports leads to a solution.
I suspect that making swap large enough will solve it (although it may get very slow).

popcornmix, 2015-02-27:
Closing as I don't believe a firmware/kernel update can fix this.

closed 2015-02-27 by popcornmix