Hexxeh/rpi-firmware #77: with actual firmware no mpeg2 possible

jsffm, 2015-02-26:
back  to ff93337300068d3cddd7f823bc77c4eb954357c9 everything ok

popcornmix, 2015-02-27:
Need more info. I've played mpeg2 files with latest firmware and kodi.
Can you explain exactly what you ran, and ideally link to a file that no longer plays.

jsffm, 2015-02-27:
The problem appeared with rpihddevice http://projects.vdr-developer.org/projects/plg-rpihddevice tested with live-tv and recordings, I will also contact the maintainer of the plugin
OS is Raspian, I did an update and upgrade before updating the firmware, rpihddevice rebuild after that.

popcornmix, 2015-02-27:
Does a sample recoding play with (command line) omxplayer in raspbian?
If it doesn't (since the update) can you post a sample file (e.g. to dropbox or google drive).

jsffm, 2015-02-27:
I will test it, hardware is a Pi 2

popcornmix, 2015-02-27:
Can you also check output of
```vcgencmd codec_enabled MPG2```

jsffm, 2015-02-27:
root@raspberrypi-2:~# vcgencmd codec_enabled MPG2
MPG2=enabled

omxplayer works fine, so it is a problem of rpihddevice, I must contact the maintainer.

popcornmix, 2015-02-27:
So the firmware version is irrelevant? (i.e. downgrading firmware doesn't fix it?)
Was the problem was due to update in rpihddevice? Is so then please close the issue.

jsffm, 2015-02-27:
downgrading firmware fixed it

popcornmix, 2015-02-27:
If downgrading firmware does fix it, then we'd like to understand the problem.
We may need to involve the rpihddevice developer.

Can you explain the simplest way to reproduce the problem?
I have no live TV hardware, but can I install an application on raspbian and them try to play an MPEG2 file to see the problem?

jsffm, 2015-02-27:
You need vdr from here ftp://ftp.tvdr.de/vdr/vdr-2.2.0.tar.bz2 and rpihddevice.
For testing you need either a tv-card (DVB-T?) or a recording

jsffm, 2015-02-27:
Another user telled that it works after update, upgrade and rebuild, I did all these things with no success, maybe in wrong sequence. I have to try it again.

popcornmix, 2015-02-27:
Okay, report back if you get it working with latest firmware.

popcornmix, 2015-02-28:
Can you try latest rpi-update firmware? Includes a fix for MPEG-2 decode.

jsffm, 2015-02-28:
Sure, but you have to wait till tomorrow

realglotzi, 2015-02-28:
The issue is fixed for me. Thx.

jsffm, 2015-03-01:
mpeg2 works fine with current firmware

closed 2015-03-01 by jsffm