Hexxeh/rpi-firmware #82: mmc-bcm2835: Unable to initialise DMA channels. Falling back to PIO

misterboo, 2015-05-21:
I get this error with since some time, the error is described her:

https://github.com/raspberrypi/linux/issues/682

pelwell, 2015-05-21:
The diagnosis is the same as on that thread - a mismatch between the .dtb and kernel. We try to avoid changes that need synchronised updates, but sometimes they are necessary, and since they are both replaceed at the same time in official updates it should only affect people building their own kernels.

What is the (recent) update history of your board? Are you running NOOBS?

Have you tried:
```
sudo apt-get update
sudo apt-get upgrade
```
or
```
sudo rpi-update
```
(only choose the latter if you normally do and know what you are doing)?

misterboo, 2015-05-21:
i always update with rpi-update .... the error msg is still in the latest update

pelwell, 2015-05-21:
What do these command output?
```
dmesg | head -25
cat /etc/fstab
for a in /proc/device-tree/soc/mmc*/*; do echo $a; od -Ax -tx1 $a; done
```

misterboo, 2015-05-21:
found it ...

It happened only in the latest version

https://github.com/Hexxeh/rpi-firmware/commit/56569584e2a65d50408ea56d467336446af69925

kernel: Adding bcm2835-sdhost driver, and an overlay to enable it

The solution: Add dtoverlay=sdhost in config.txt

closed 2015-05-21 by misterboo

pelwell, 2015-05-21:
Although I'm happy to see this issue closed, I want you to understand what you've done.

"sdhost" is an alternative, experimental SD card driver (the current default is called "mmc"). As you've found, sdhost is selected by using that overlay, but in order for this to work the MMC driver must be selected based on the contents of the DTB. If your DTB is old this information won't be in there, and you'll get the error.

The sdhost overlay is effectively self contained, so it will work with an old DTB, so by using the overlay you've sidestepped the issue of the base DTB not containing the relevant info.

So far we've had no reports of problems using sdhost, but be aware that we haven't approved it for everyday use yet.

misterboo, 2015-05-21:
It has surely to to with the latest commit. As an alternative i was going back to the last firmware version. That also works. The error was only in the latest firmware with some commits from popcornmix 

https://github.com/Hexxeh/rpi-firmware/commit/03b44377dd1f410900d93ca92d39cdacba3544ac

uecasm, 2015-06-09:
I had the same message, and adding `dtoverlay=sdhost` also made it disappear for me; I don't know if there are any other consequences of this but it still appears to be working ok.

I did wonder, given the other overlay files present, whether adding `dtoverlay=mmc` instead would also work, but while this does not appear to cause any harm it does not remove the warning.

uecasm, 2015-06-10:
Actually, never mind.  I see https://github.com/Hexxeh/rpi-firmware/commit/a1621d67b20f04acdac602d0b24923ec3cdbfb66 addresses this, and I can confirm that upgrading to firmware https://github.com/raspberrypi/firmware/commit/4d961a664d4b7d resolves this without using `dtoverlay=sdhost`.  Apologies for the noise.