Hexxeh/rpi-firmware #83: Pi Cam stopped working after upgrading to actual kernel (2015-05-30)

cstueben, 2015-05-31:
Pi Cam stopped working after upgrading to actual kernel (2015-05-30) on rpi model b

pelwell, 2015-05-31:
Sorry about that, and thanks for the report - I think I know why that might be.

Can you post the output of `dmesg | head -25`?

Until there is a fix, you can revert to the previous version using:
```
sudo rpi-update 4f95c894c6a96b265d98f04a889e469833f7e996
```

cstueben, 2015-05-31:
This is the output of ```dmesg | head -25```

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.14+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #793 PREEMPT Sat May 30 13:15:19 BST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B Plus Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0845534, node_mem_map d74f4000
[    0.000000]   Normal zone: 768 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 97536
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708.boardrev=0x5 bcm2708.serial=0xb9b217ca smsc95xx.macaddr=B8:27:EB:B2:17:CA bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 372216K/393216K available (5902K kernel code, 348K rwdata, 1876K rodata, 340K init, 733K bss, 21000K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
```

pelwell, 2015-05-31:
A fix has been pushed. Please update and retry when convenient.

cstueben, 2015-05-31:
I applied the patch successfully to my rpi. The cam works fine again.

closed 2015-05-31 by cstueben