Hexxeh/rpi-firmware #86: firmware bitrate header breaks gst use

cangaru, 2015-06-28:
I get this error using gstreamer in the latest firmware and it wasn't a problem in any firmware prior to that: 0x13b96c0 ERROR            omxvideoenc gstomxvideoenc.c:296:gst_omx_video_enc_open:<omxh264enc-omxh264enc0> Failed to set bitrate parameters: Bad port index (0x8000101b)

popcornmix, 2015-06-28:
@6by9 I assume this is caused by "[PATCH] video_encode: Initialise headers for video_bitrate"

6by9, 2015-06-28:
Drat, noticed nSize and nVersion referenced in https://github.com/raspberrypi/firmware/issues/163, but missed port. Fixing the 2 probably bypasses any workaround that was added. 
One extra line needed in the same place as the other two:
```   port->video_bitrate.nPortIndex = nPortIndex;```

6by9, 2015-06-28:
@cangaru Which version of Gstreamer is being run here, and where is the appropriate source? I'd ideally like to check for any workarounds- a quick search for gstomxvideoenc.c brings up multiple hits, but the ones I've checked don't have that error message on line 296.

cangaru, 2015-06-28:
it is gstreamer 1.2 as from the raspbian repository.

cangaru, 2015-07-01:
Looks good. thanks.

closed 2015-07-01 by popcornmix