Hexxeh/rpi-firmware #88: gstreamer gets stuck

cangaru, 2015-07-23:
Latest rpi firmware, gstreamer now gets stuck. I have the last bit of the debug output, but I'm not sure what the error is:

0:00:31.520362250  2807  0x1c55c00 INFO              GST_STATES gstelement.c:2233:_priv_gst_element_state_changed:<h264parse0> notifying about state-changed NULL to READY (VOID_PENDING pending)
0:00:31.520563447  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:312:gst_bus_post:<bus1> [msg 0x1dc5418] posting on bus state-changed message: 0x1dc5418, time 99:99:99.999999999, seq-num 50, element 'h264parse0', GstMessageStateChanged, old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY, pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:31.520978705  2807  0x1c55c00 DEBUG                    bin gstbin.c:3396:gst_bin_handle_message_func:<pipeline0> [msg 0x1dc5418] handling child h264parse0 message of type state-changed
0:00:31.521166621  2807  0x1c55c00 DEBUG                    bin gstbin.c:3729:gst_bin_handle_message_func:<pipeline0> posting message upward
0:00:31.521275162  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:312:gst_bus_post:<bus2> [msg 0x1dc5418] posting on bus state-changed message: 0x1dc5418, time 99:99:99.999999999, seq-num 50, element 'h264parse0', GstMessageStateChanged, old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY, pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:31.521573754  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:347:gst_bus_post:<bus2> [msg 0x1dc5418] pushing on async queue
0:00:31.521834481  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:350:gst_bus_post:<bus2> [msg 0x1dc5418] pushed on async queue
0:00:31.522003230  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:343:gst_bus_post:<bus1> [msg 0x1dc5418] dropped
0:00:31.522115782  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2563:gst_element_set_state_func:<h264parse0> returned SUCCESS
0:00:31.522430467  2807  0x1c55c00 INFO              GST_STATES gstbin.c:2656:gst_bin_change_state_func:<pipeline0> child 'h264parse0' changed state to 2(READY) successfully
0:00:31.522727341  2807  0x1c55c00 DEBUG                    bin gstbin.c:2094:gst_bin_sort_iterator_next:<pipeline0> queue head gives ac3parse0
0:00:31.522837913  2807  0x1c55c00 DEBUG                    bin gstbin.c:885:find_message:<pipeline0> no message found matching types 00001000
0:00:31.522991506  2807  0x1c55c00 DEBUG                    bin gstbin.c:892:find_message:<pipeline0>   structure-change
0:00:31.523150515  2807  0x1c55c00 DEBUG                    bin gstbin.c:2010:update_degree:<pipeline0> change element avenc_ac3-0, degree 1->0, linked to ac3parse0
0:00:31.523430826  2807  0x1c55c00 DEBUG                    bin gstbin.c:1893:add_to_queue:<pipeline0> adding 'avenc_ac3-0' to queue
0:00:31.523723376  2807  0x1c55c00 INFO              GST_STATES gstbin.c:2227:gst_bin_element_set_state:<ac3parse0> current NULL pending VOID_PENDING, desired next READY
0:00:31.523904261  2807  0x1c55c00 DEBUG                    bin gstbin.c:885:find_message:<pipeline0> no message found matching types 00100000
0:00:31.524014208  2807  0x1c55c00 DEBUG                    bin gstbin.c:892:find_message:<pipeline0>   async-start
0:00:31.524333894  2807  0x1c55c00 DEBUG                    bin gstbin.c:2322:gst_bin_element_set_state:<pipeline0> setting element ac3parse0 to READY, base_time 0:00:00.000000000
0:00:31.524617486  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2479:gst_element_set_state_func:<ac3parse0> set_state to READY
0:00:31.524729985  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2504:gst_element_set_state_func:<ac3parse0> setting target state to READY
0:00:31.524885401  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2517:gst_element_set_state_func:<ac3parse0> current NULL, old_pending VOID_PENDING, next VOID_PENDING, old return SUCCESS
0:00:31.525064723  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2551:gst_element_set_state_func:<ac3parse0> final: setting state from NULL to READY
0:00:31.525348315  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2636:gst_element_change_state:<ac3parse0> element changed state SUCCESS
0:00:31.525626386  2807  0x1c55c00 INFO              GST_STATES gstelement.c:2328:gst_element_continue_state:<ac3parse0> completed state change to READY
0:00:31.525782375  2807  0x1c55c00 INFO              GST_STATES gstelement.c:2233:_priv_gst_element_state_changed:<ac3parse0> notifying about state-changed NULL to READY (VOID_PENDING pending)
0:00:31.525982790  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:312:gst_bus_post:<bus1> [msg 0x1dc5380] posting on bus state-changed message: 0x1dc5380, time 99:99:99.999999999, seq-num 51, element 'ac3parse0', GstMessageStateChanged, old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY, pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:31.526353726  2807  0x1c55c00 DEBUG                    bin gstbin.c:3396:gst_bin_handle_message_func:<pipeline0> [msg 0x1dc5380] handling child ac3parse0 message of type state-changed
0:00:31.526477319  2807  0x1c55c00 DEBUG                    bin gstbin.c:3729:gst_bin_handle_message_func:<pipeline0> posting message upward
0:00:31.526621537  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:312:gst_bus_post:<bus2> [msg 0x1dc5380] posting on bus state-changed message: 0x1dc5380, time 99:99:99.999999999, seq-num 51, element 'ac3parse0', GstMessageStateChanged, old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY, pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:31.527042472  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:347:gst_bus_post:<bus2> [msg 0x1dc5380] pushing on async queue
0:00:31.527156534  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:350:gst_bus_post:<bus2> [msg 0x1dc5380] pushed on async queue
0:00:31.527314658  2807  0x1c55c00 DEBUG                GST_BUS gstbus.c:343:gst_bus_post:<bus1> [msg 0x1dc5380] dropped
0:00:31.527468563  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2563:gst_element_set_state_func:<ac3parse0> returned SUCCESS
0:00:31.527764968  2807  0x1c55c00 INFO              GST_STATES gstbin.c:2656:gst_bin_change_state_func:<pipeline0> child 'ac3parse0' changed state to 2(READY) successfully
0:00:31.528033351  2807  0x1c55c00 DEBUG                    bin gstbin.c:2094:gst_bin_sort_iterator_next:<pipeline0> queue head gives capsfilter1
0:00:31.528145538  2807  0x1c55c00 DEBUG                    bin gstbin.c:885:find_message:<pipeline0> no message found matching types 00001000
0:00:31.528299808  2807  0x1c55c00 DEBUG                    bin gstbin.c:892:find_message:<pipeline0>   structure-change
0:00:31.528454547  2807  0x1c55c00 DEBUG                    bin gstbin.c:2010:update_degree:<pipeline0> change element omxh264enc-omxh264enc0, degree 1->0, linked to capsfilter1
0:00:31.528738868  2807  0x1c55c00 DEBUG                    bin gstbin.c:1893:add_to_queue:<pipeline0> adding 'omxh264enc-omxh264enc0' to queue
0:00:31.529031314  2807  0x1c55c00 INFO              GST_STATES gstbin.c:2227:gst_bin_element_set_state:<capsfilter1> current NULL pending VOID_PENDING, desired next READY
0:00:31.529210480  2807  0x1c55c00 DEBUG                    bin gstbin.c:885:find_message:<pipeline0> no message found matching types 00100000
0:00:31.529319906  2807  0x1c55c00 DEBUG                    bin gstbin.c:892:find_message:<pipeline0>   async-start
0:00:31.529613134  2807  0x1c55c00 DEBUG                    bin gstbin.c:2322:gst_bin_element_set_state:<pipeline0> setting element capsfilter1 to READY, base_time 0:00:00.000000000
0:00:31.529894226  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2479:gst_element_set_state_func:<capsfilter1> set_state to READY
0:00:31.530007090  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2504:gst_element_set_state_func:<capsfilter1> setting target state to READY
0:00:31.530161047  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2517:gst_element_set_state_func:<capsfilter1> current NULL, old_pending VOID_PENDING, next VOID_PENDING, old return SUCCESS
0:00:31.530335473  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2551:gst_element_set_state_func:<capsfilter1> final: setting state from NULL to READY
0:00:31.530634586  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2636:gst_element_change_state:<capsfilter1> element changed state SUCCESS
[... cut ...]
0:00:31.540489216  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2504:gst_element_set_state_func:<omxh264enc-omxh264enc0> setting target state to READY
0:00:31.540750725  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2517:gst_element_set_state_func:<omxh264enc-omxh264enc0> current NULL, old_pending VOID_PENDING, next VOID_PENDING, old return SUCCESS
0:00:31.541061349  2807  0x1c55c00 DEBUG             GST_STATES gstelement.c:2551:gst_element_set_state_func:<omxh264enc-omxh264enc0> final: setting state from NULL to READY

popcornmix, 2015-07-23:
Could you test this firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_150724.zip

unzip into boot directory of sdcard

cangaru, 2015-07-24:
Currently transcoding, when it finishes I will try it. Thanks

https://dl.dropboxusercontent.com/u/3669512/temp/firmware_150724.zip

cangaru, 2015-07-24:
looks like it's working ok.
only tested one video so far...
thanks
John

popcornmix, 2015-07-24:
This fix should be in latest rpi-update firmware

cangaru, 2015-07-24:
Nope: updated latest firmware and not working...

0:00:05.562282818  2685   0x26bc00 INFO              GST_STATES
gstbin.c:2656:gst_bin_change_state_func:<pipeline0> child 'ac3parse0'
changed state to 2(READY) successfully
0:00:05.562404902  2685   0x26bc00 DEBUG                    bin
gstbin.c:2094:gst_bin_sort_iterator_next:<pipeline0> queue head gives
capsfilter1
0:00:05.562476933  2685   0x26bc00 DEBUG                    bin
gstbin.c:885:find_message:<pipeline0> no message found matching types
00001000
0:00:05.562573548  2685   0x26bc00 DEBUG                    bin
gstbin.c:892:find_message:<pipeline0>   structure-change
0:00:05.562670632  2685   0x26bc00 DEBUG                    bin
gstbin.c:2010:update_degree:<pipeline0> change element
omxh264enc-omxh264enc0, degree 1->0, linked to capsfilter1
0:00:05.562801935  2685   0x26bc00 DEBUG                    bin
gstbin.c:1893:add_to_queue:<pipeline0> adding 'omxh264enc-omxh264enc0'
to queue
0:00:05.562906102  2685   0x26bc00 INFO              GST_STATES
gstbin.c:2227:gst_bin_element_set_state:<capsfilter1> current NULL
pending VOID_PENDING, desired next READY
0:00:05.563019487  2685   0x26bc00 DEBUG                    bin
gstbin.c:885:find_message:<pipeline0> no message found matching types
00100000
0:00:05.563087300  2685   0x26bc00 DEBUG                    bin
gstbin.c:892:find_message:<pipeline0>   async-start
0:00:05.563438343  2685   0x26bc00 DEBUG                    bin
gstbin.c:2322:gst_bin_element_set_state:<pipeline0> setting element
capsfilter1 to READY, base_time 0:00:00.000000000
0:00:05.563562979  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2479:gst_element_set_state_func:<capsfilter1> set_state to
READY
0:00:05.563662302  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2504:gst_element_set_state_func:<capsfilter1> setting
target state to READY
0:00:05.563762459  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2517:gst_element_set_state_func:<capsfilter1> current NULL,
old_pending VOID_PENDING, next VOID_PENDING, old return SUCCESS
0:00:05.563846053  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2551:gst_element_set_state_func:<capsfilter1> final:
setting state from NULL to READY
0:00:05.563951418  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2636:gst_element_change_state:<capsfilter1> element changed
state SUCCESS
0:00:05.564145533  2685   0x26bc00 INFO              GST_STATES
gstelement.c:2328:gst_element_continue_state:<capsfilter1> completed
state change to READY
0:00:05.564246054  2685   0x26bc00 INFO              GST_STATES
gstelement.c:2233:_priv_gst_element_state_changed:<capsfilter1>
notifying about state-changed NULL to READY (VOID_PENDING pending)
0:00:05.564375951  2685   0x26bc00 DEBUG                GST_BUS
gstbus.c:312:gst_bus_post:<bus1> [msg 0x3db328] posting on bus
state-changed message: 0x3db328, time 99:99:99.999999999, seq-num 52,
element 'capsfilter1', GstMessageStateChanged,
old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY,
pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:05.564548660  2685   0x26bc00 DEBUG                    bin
gstbin.c:3396:gst_bin_handle_message_func:<pipeline0> [msg 0x3db328]
handling child capsfilter1 message of type state-changed
0:00:05.564628556  2685   0x26bc00 DEBUG                    bin
gstbin.c:3729:gst_bin_handle_message_func:<pipeline0> posting message upward
0:00:05.564743817  2685   0x26bc00 DEBUG                GST_BUS
gstbus.c:312:gst_bus_post:<bus2> [msg 0x3db328] posting on bus
state-changed message: 0x3db328, time 99:99:99.999999999, seq-num 52,
element 'capsfilter1', GstMessageStateChanged,
old-state=(GstState)GST_STATE_NULL, new-state=(GstState)GST_STATE_READY,
pending-state=(GstState)GST_STATE_VOID_PENDING;
0:00:05.564956474  2685   0x26bc00 DEBUG                GST_BUS
gstbus.c:347:gst_bus_post:<bus2> [msg 0x3db328] pushing on async queue
[... 100 lines cut ...]
00100000
0:00:05.569667845  2685   0x26bc00 DEBUG                    bin
gstbin.c:892:find_message:<pipeline0>   async-start
0:00:05.569759720  2685   0x26bc00 DEBUG                    bin
gstbin.c:2322:gst_bin_element_set_state:<pipeline0> setting element
omxh264enc-omxh264enc0 to READY, base_time 0:00:00.000000000
0:00:05.569899460  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2479:gst_element_set_state_func:<omxh264enc-omxh264enc0>
set_state to READY
0:00:05.569969408  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2504:gst_element_set_state_func:<omxh264enc-omxh264enc0>
setting target state to READY
0:00:05.570062638  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2517:gst_element_set_state_func:<omxh264enc-omxh264enc0>
current NULL, old_pending VOID_PENDING, next VOID_PENDING, old return
SUCCESS
0:00:05.570351754  2685   0x26bc00 DEBUG             GST_STATES
gstelement.c:2551:gst_element_set_state_func:<omxh264enc-omxh264enc0>
final: setting state from NULL to READY
0:00:05.628540558  2685   0x26bc00 DEBUG                    omx
gstomx.c:129:gst_omx_core_acquire: Successfully loaded core
'/opt/vc/lib/libopenmaxil.so'
0:00:05.629315248  2685   0x26bc00 DEBUG                    omx
gstomx.c:144:gst_omx_core_acquire: Successfully initialized core
'/opt/vc/lib/libopenmaxil.so'

popcornmix, 2015-07-24:
Does the Dropbox test firmware work in this case?

cangaru, 2015-07-25:
I'll test again

cangaru, 2015-07-26:
Yes, it does work with the zip I downloaded before...
J

popcornmix, 2015-07-26:
Can you provide something I can run to see the failure?
E.g. tar up your gstreamer and video file? Binary only is okay.

cangaru, 2015-07-26:
The video is very big. I'll have a try tomorrow, OK?

popcornmix, 2015-07-26:
Okay. Try a few videos and use the smallest one (or edit it down with ffmpeg). If necessary I can download a large file.

cangaru, 2015-07-27:
There is another possibility. If, while it is transcoding one video it
gets a request to do another, it says insufficient resources and then
the first transcode job freezes. If I kill it, it does not free up
whatever it is, and it will not do any more jobs until I do a reboot.
This is also not good behaviour, though. That may also be the case with
the latest rpi-update, but I haven't checked.
J

cangaru, 2015-07-28:
Is there a problem with the sample I sent you?
john

popcornmix, 2015-07-28:
Not received anything. Did you send to popcornmix at gmail dot com ?

popcornmix, 2015-07-28:
I have an update to the firmware that will be pushed soon. Can you test it:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_timestamp1.zip

cangaru, 2015-07-28:
no, to notifications... sorry... try again...

cangaru, 2015-07-28:
Seems to be good.

cangaru, 2015-07-29:
Second test failed, also stuck....

0:00:00.802189896  2787 0x6fd03400 INFO               GST_EVENT
gstevent.c:628:gst_event_new_caps: creating caps event video/x-raw,
format=(string)RGBx, width=(int)1440, height=(int)1080,
pixel-aspect-ratio=(fraction)4/3, interlace-mode=(string)mixed,
colorimetry=(string)1:1:0:0, framerate=(fraction)25/1
0:00:00.843063074  2787 0x6fd03400 INFO                     omx
gstomx.c:1864:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Setting resize port 60 to disabled
0:00:00.844233277  2787 0x6fd03400 INFO                     omx
gstomx.c:1912:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Set resize port 60 to disabled: None (0x00000000)
0:00:00.844739264  2787 0x6fd03400 INFO                     omx
gstomx.c:2153:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Waiting for resize port 60 to be disabled
0:00:00.845416865  2787 0x6fd03400 INFO                     omx
gstomx.c:2228:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
resize port 60 is disabled: None (0x00000000)
0:00:00.845789415  2787 0x6fd03400 INFO                     omx
gstomx.c:1864:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Setting resize port 61 to disabled
0:00:00.846627067  2787 0x6fd03400 INFO                     omx
gstomx.c:1912:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Set resize port 61 to disabled: None (0x00000000)
0:00:00.847057742  2787 0x6fd03400 INFO                     omx
gstomx.c:2153:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Waiting for resize port 61 to be disabled
0:00:00.847671541  2787 0x6fd03400 INFO                     omx
gstomx.c:2228:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
resize port 61 is disabled: None (0x00000000)
0:00:00.848312058  2787 0x6fd03400 INFO                     omx
gstomx.c:1864:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Setting resize port 60 to enabled
0:00:00.849149867  2787 0x6fd03400 INFO                     omx
gstomx.c:1912:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Set resize port 60 to enabled: None (0x00000000)
0:00:00.849524500  2787 0x6fd03400 INFO                     omx
gstomx.c:756:gst_omx_component_set_state:<omxmpeg2videodec-omxmpeg2videodec0>
Setting resize state from Loaded to Idle
0:00:00.850409496  2787 0x6fd03400 INFO                     omx
gstomx.c:225:gst_omx_component_handle_messages:<omxmpeg2videodec-omxmpeg2videodec0>
resize state change to Idle finished
0:00:00.850782567  2787 0x6fd03400 INFO                     omx
gstomx.c:2153:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Waiting for resize port 60 to be enabled
0:00:00.851519750  2787 0x6fd03400 INFO                     omx
gstomx.c:2228:gst_omx_port_wait_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
resize port 60 is enabled: None (0x00000000)
0:00:00.860554028  2787 0x6fd03400 INFO                     omx
gstomx.c:1864:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Setting resize port 61 to enabled
0:00:00.861443971  2787 0x6fd03400 INFO                     omx
gstomx.c:1912:gst_omx_port_set_enabled_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Set resize port 61 to enabled: None (0x00000000)
0:00:00.861848188  2787 0x6fd03400 INFO                     omx
gstomx.c:1631:gst_omx_port_allocate_buffers_unlocked:<omxmpeg2videodec-omxmpeg2videodec0>
Allocating 4 buffers of size 6266880 for resize port 61

cangaru, 2015-07-29:
sent now...

Ruffio, 2016-07-24:
@cangaru do you still experience this issue? If not, please close this issue. Thanks.

Ruffio, 2017-01-22:
@popcornmix accorging to https://github.com/Hexxeh/rpi-firmware/issues/88#issuecomment-125751965 this looks like it can be closed

Ruffio, 2017-03-03:
@popcornmix this issue should be closed due to lack of response from originator. Originator has been pinged three times within the last half a year without any response...

closed 2017-03-03 by popcornmix