Hexxeh/rpi-firmware #92: HDMI to DVI output possibly broken

dhowdy, 2015-09-01:
Hello,

I've found that using a passive HDMI to DVI adapter/cable does not work on the Pi 2 starting at commit 03b44377dd1f410900d93ca92d39cdacba3544ac (May 13, 2015).  I've tested this across multiple Pi 2s, montiors, and cables/adapters.  The behavior that I see is that omxplayer is able to display output video, but nothing else (including the X session) is visible.  If I use the previous commit everything works just fine.

Hot-swapping monitors yields the same results; HDMI works, DVI does not. 

I've tried about every combination of hdmi_drive, hmdi_group, hdmi_mode, hdmi_boost and the like to attempt to work around this issue.  Right now the only thing that I've found that works is using `rpi-update 
8521fd34c8f66b6d109acce943f6e25ec93ec005` 

Thanks.

popcornmix, 2015-09-01:
Can you create this here:
https://github.com/raspberrypi/firmware/issues

This is the wrong place and it's likely to get lost here.

closed 2015-09-01 by popcornmix

dhowdy, 2015-09-01:
Sorry about that!