Hexxeh/rpi-firmware #94: USB-Harddrive gets fully powered down

lduesing, 2015-10-09:
in 4.1.10 my USB connected harddrive gets fully powered down from time to time. Due to being there my OS my raspberry freezes complete. Interestingly only the HDD connected via "JM20329 SATA Bridge" is powered down. A "Western Digital Technologies, Inc. Elements 2TB" does not show this issue as far as I can see.

rickysarraf, 2015-10-09:
On Fri, 2015-10-09 at 05:18 -0700, lduesing wrote:
> in 4.1.10 my USB connected harddrive gets fully powered down from
> time to time. Due to being there my OS my raspberry freezes complete.
> Interestingly only the HDD connected via "JM20329 SATA Bridge" [...]

I believe I've recently started running into the same problem. I am
thankful that you spent the time to root cause it.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

popcornmix, 2015-10-09:
Not really the right place for this. If you think this is a kernel bug the report here: https://github.com/raspberrypi/linux/
But if it's a general discussion then the forums would be best.
Must say, if you power off a hard drive that contains the OS, then I would expect a hang.

closed 2015-10-09 by popcornmix