Hexxeh/rpi-firmware #95: We want kernal 4.2 for MT7601 drivers

MaoJianwei, 2015-11-10:
We want kernal 4.2 for MT7601 drivers

volschin, 2015-11-10:
Check out this description hör to integrate 
http://groenholdt.net/Computers/RaspberryPi/MediaTek-MT7601-USB-WIFI-on-the-Raspberry-Pi/MediaTek-MT7601-USB-WIFI-on-the-Raspberry-Pi.html

skvatboy, 2015-11-14:
The MT7601 drivers is in 4.1.5+ 
https://github.com/Hexxeh/rpi-firmware/commit/c9238d33ef3d7bcc6b20ae34419825666cfdb89d
add the mt7601u.bin 
sudo wget https://github.com/porjo/mt7601/raw/master/src/mcu/bin/MT7601.bin -O /lib/firmware/mt7601u.bin

MaoJianwei, 2015-11-14:
thank you two friends very much! 
I will have a try tomorrow!

Ruffio, 2016-06-30:
@MaoJianwei is this PR still relevant?

Ruffio, 2016-07-26:
@MaoJianwei did it work?

popcornmix, 2016-07-26:
Closing anyway. We are now on 4.4 kernel so this issue is no longer relevant.

closed 2016-07-26 by popcornmix