Hexxeh/rpi-firmware #96: Backlight toggle function for official raspberry pi DSI display

Snyder1977, 2015-11-25:
The function to turn on/off the backlight of my pi is unstable. I could imagine that it is related to the firmware but I am not sure and do not know where to dig deeper.

I usually run the following commands:

To turn the Display on:
 'sudo bash -c "echo 0 > /sys/class/backlight/rpi_backlight/bl_power"' 

 To turn the Display off:
 'sudo bash -c "echo 1 > /sys/class/backlight/rpi_backlight/bl_power"'

Sometimes it is working fine for me and sometimes it crashes the pi. The pi crashes and reboots immediatley after the execution one of the above mentioned commands.

I don´t know what the reason might be? Do you have any ideas how to fix it?

For further details please have a look at this thread:

https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=120968&sid=3914e9536acc702e0d27d490e2f6cc8e&start=75

popcornmix, 2015-11-25:
Report the issue here: https://github.com/raspberrypi/linux/issues
where the right developer might see it.

closed 2015-11-25 by popcornmix

Snyder1977, 2015-11-25:
ok, many thanks for pointing out to the right place.