Hexxeh/rpi-firmware #98: Problem reading serial USB port with version 4.1.15+

Menox10, 2015-12-22:
With this version I have problems with reading to open /dev/ttyUSB0. Can not read data
anymore. It is solved with a roll-back to version 4.1.14

lsusb
Bus 001 Device 004: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port

popcornmix, 2015-12-23:
Can you confirm that https://github.com/Hexxeh/rpi-firmware/commit/17cf280ac76df9d9c3d1305709cc2d74fbc371b9 fails and https://github.com/Hexxeh/rpi-firmware/commit/98e801605b31a5d7a4c36de5cb1d0e33ce95e8ab is okay?
The only change between those two points was upstream kernel changes.

Menox10, 2015-12-24:
> Can you confirm that 17cf280
> fails and 98e8016

This version in running fine 17cf280

root@powerpi ~ # uname -a
Linux powerpi 4.1.14-v7+ #828 SMP PREEMPT Thu Dec 10 13:22:08 GMT 2015
armv7l GNU/Linux

After update to latest version I got problems with USB serial connection
(tried to times). The scripts I use
to read the smartmeter is running well for many years (no script error).

I also discovered that the Raspberry Pi is running on multiple kernel
versions. When I remove all kernel
version expect the most recent in /lib/modules the WIFI connection will not
start after reboot. I noticed that
it tried to startup something from a previous (now removed) kernel version.
This is not correct. It should
be sufficient to only keep 4.1.15+ and 4.1.15-v7+.

The only problem I have with this cheap USB-RS232 adapter is that it uses
the PL2303x chips (less memory).
The pl2303.ko file is not 100% compatible with this chip. This causes
sometimes the ETH0 port to stop and reboot is necessary.
Maybe it is related with the problem in the latest kernel version??
Hope you or a team member can make it compatible.

This results in the kernel error /var/log/syslog
Dec 23 15:38:14 powerpi kernel: [98399.788973] pl2303 ttyUSB0: error
sending break = -32
Dec 23 15:40:01 powerpi kernel: [98507.338124] pl2303 ttyUSB0:
pl2303_get_line_request - failed: -32

root@powerpi /lib/modules # ll
total 32
drwxr-xr-x  8 root root 4096 Dec 22 12:14 .
drwxr-xr-x 14 root root 4096 May  7  2015 ..
drwxr-xr-x  3 root root 4096 Dec 21 11:55 4.1.13+
drwxr-xr-x  3 root root 4096 Dec 21 11:55 4.1.13-v7+
drwxr-xr-x  3 root root 4096 Dec 22 12:14 4.1.14+
drwxr-xr-x  3 root root 4096 Dec 22 12:14 4.1.14-v7+
drwxr-xr-x  3 root root 4096 Dec 22 11:25 4.1.15+
drwxr-xr-x  3 root root 4096 Dec 22 11:25 4.1.15-v7+

Menox10, 2015-12-25:
Tonight I had again a crash (/var/log/syslog). The link beat get lost and
no eth0 connection anymore

Dec 25 08:58:01 powerpi kernel: [247188.152189] pl2303 ttyUSB0:
pl2303_get_line_request - failed: -32
Dec 25 08:58:13 powerpi kernel: [247200.561558] pl2303 ttyUSB0: error
sending break = -32
Dec 25 08:58:14 powerpi kernel: [247200.994083] smsc95xx 1-1.1:1.0 eth0:
Failed to read reg index 0x00000114: -71
Dec 25 08:58:14 powerpi kernel: [247200.994106] smsc95xx 1-1.1:1.0 eth0:
Error reading MII_ACCESS
Dec 25 08:58:14 powerpi kernel: [247200.994118] smsc95xx 1-1.1:1.0 eth0:
MII is busy in smsc95xx_mdio_read
Dec 25 08:58:14 powerpi ifplugd(eth0)[1509]: Link beat lost.

I tried again with the latest kernel version

root@powerpi ~ # uname -a
Linux powerpi 4.1.15-v7+ #830 SMP Tue Dec 15 17:02:45 GMT 2015 armv7l
GNU/Linux

It seems to work but much slower in response. I still have the old errors

Dec 25 10:41:26 powerpi kernel: [   71.923211] pl2303 ttyUSB0:
pl2303_get_line_request - failed: -32
Dec 25 10:41:26 powerpi kernel: [   71.924576] pl2303 ttyUSB0:
pl2303_get_line_request - failed: -32
Dec 25 10:41:40 powerpi kernel: [   85.086759] pl2303 ttyUSB0: error
sending break = -32
Dec 25 10:42:02 powerpi kernel: [  107.111505] pl2303 ttyUSB0:
pl2303_get_line_request - failed: -32
Dec 25 10:42:14 powerpi kernel: [  119.897088] pl2303 ttyUSB0: error
sending break = -32
^C
root@powerpi ~ # P1uitlezer.py
('DSMR P1 uitlezer', '2.1')
Control-C om te stoppen
Poort: (/dev/ttyUSB0)
daldag                   5120  kWh
piekdag                  6241  kWh
dalterug                 1033  kWh
piekterug                2539  kWh
daltotaal                4087  kWh
piektotaal               3702  kWh
totaal                   7789  kWh
afgenomen vermogen       20  W
teruggeleverd vermogen   0  W
Gas                      4352065  dm3
Datum                    151225
Tijd                     1000
daldag                   5120  kWh
piekdag                  6241  kWh

Hope this can be solved in future versions.

Thanks
Menno

2015-12-24 19:23 GMT+01:00 Menno Norden <menno.norden@gmail.com>:

Menox10, 2016-01-03:
I found a solution for the PL2303x driver for Ubuntu. Maybe someone can also use this to create it for Debian wheezy.

http://koti.mbnet.fi/lonnberg/pl2303x.html

popcornmix, 2016-01-03:
That patch looks to be designed for 2.6 kernel. It doesn't apply to 4.x kernels we use.
Anyway, if the patch is good it needs to be submitted upstream where we can cherry-pick it.
We avoid any patches that haven't been accepted upstream, and so haven't been peer reviewed by the experts.

Menox10, 2016-01-03:
I think the patch is not okay to use as it is, but the idea behind it van
be used. I think some C-code needs to be modified and compiled to support
both types.

I am not an expert in C or compile kernel code.

I am available for testing new code

Thanks
Menno

Ruffio, 2016-07-24:
@Menox10 have you raised your issue upstream? RPi has been running kernal v. 4.x for some time now and I don't think that Wheezy is supported any longer on RPI, only Jessie. If the issue persists with Jessie please raise the issue upstream. Consider closing this issue.

Ruffio, 2017-03-03:
@popcornmix this issue should be closed due to lack of response. Originator has been pinged twice with half a year in between...

closed 2017-03-03 by popcornmix