Hexxeh/rpi-firmware #102: issue in playing video using omxplayer (firmware related?)

navaneetharaopy, 2016-01-21:
**Initial Setup** 
*OS : Debian OS jessie 
Board: raspberry pi 2 model B
Firmware Version: `Linux raspberrypi 4.1.15-v7+ #831 SMP Tue Jan 19 18:39:46 GMT 2016 armv7l GNU/Linux`
command used: `sudo omxplayer --orientation 90 --win '0 0 1280 720' -o both  test\ 30fps.mp4`*

Played below attached video(test 30fps.mp4) and found glitch while playing. So i downgraded using command  **`sudo rpi-update 65aad25162cc39e61d104d87af8ff46cb81308da`**

**After Downgrade**
 *OS : Debian OS jessie 
Board: raspberry pi 2 model B
Firmware Version: `Linux raspberrypi 4.1.12-v7+ #824 SMP PREEMPT Wed Oct 28 16:46:35 GMT 2015 armv7l GNU/Linux`
command used: `sudo omxplayer --orientation 90 --win '0 0 1280 720' -o both test\ 30fps.mp4`*

Played same video once again and found NO glitch. Video play was smooth.
[test 30fps.mp4.zip](https://github.com/Hexxeh/rpi-firmware/files/99070/test.30fps.mp4.zip)

popcornmix, 2016-01-21:
Duplicate of https://github.com/raspberrypi/firmware/issues/520

closed 2016-01-21 by popcornmix

nenadlazovic, 2016-03-07:
Hi popcornmix,
thanks a lot for supporting RPI2 GPU.

I did post comment on https://github.com/raspberrypi/firmware/issues/520. Pls read.

Is there a chance for enabling 1080x1920 support (broken by https://github.com/Hexxeh/rpi-firmware/commit/2b42a481e0b31cdbf2055a1c19bb2fde2a862434), so that we can have benefits of threadlock bug fix and other fixes available in newer releases ?

Thanks.

Nenad