Hexxeh/rpi-firmware #103: UNKNOWN(0x2003) SCSI Error

rickysarraf, 2016-02-10:
I've recently staretd seeing this error with my USB attached SATA 3.5" HDD. The device has been under use for more than a year. This error has been seen very recently. My guess, given the error code, and what I've understood from the conversation on debian-user [1], this could be a generic problem than being a hardware malfunction.

```
[62711.477903] sd 0:0:0:0: [sda] UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x00
[62711.485701] sd 0:0:0:0: [sda] CDB: opcode=0x28 28 00 9d 40 01 47 00 00 08 00
[62711.492910] blk_update_request: I/O error, dev sda, sector 2638217543
[84370.313684] sd 0:0:0:0: [sda] UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x00
[84370.321532] sd 0:0:0:0: [sda] CDB: opcode=0x28 28 00 a1 40 01 47 00 00 08 00
[84370.328721] blk_update_request: I/O error, dev sda, sector 2705326407
```

Reporting here hoping the latter.

[1] https://lists.debian.org/debian-user/2016/02/msg00194.html

rickysarraf, 2016-02-10:
I forgot to mention that I had also run a full fsck on the device, with no errors reported.

```
130 pi@pi:~$ sudo fsck -cvkv /dev/sdb1 
[sudo] password for pi: 
fsck from util-linux 2.25.2
e2fsck 1.42.12 (29-Aug-2014)
Checking for bad blocks (read-only test): done                                                 
SEAGATE: Updating bad block inode.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information

SEAGATE: ***** FILE SYSTEM WAS MODIFIED *****

       20013 inodes used (0.02%, out of 122101760)
        8944 non-contiguous files (44.7%)
          53 non-contiguous directories (0.3%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 16695/3216/62
   416471936 blocks used (85.28%, out of 488378008)
           0 bad blocks
          63 large files

       17086 regular files
        2885 directories
           0 character device files
           0 block device files
           0 fifos
           0 links
          33 symbolic links (32 fast symbolic links)
           0 sockets
------------
       20004 files
1 pi@pi:~$ 

```

popcornmix, 2016-02-10:
This repo is just used for hosting binaries. Bug reports should be posted here:
https://github.com/raspberrypi/linux/issues

closed 2016-02-10 by popcornmix