Hexxeh/rpi-firmware #109: 4.1.18 breaks luks encryption ("Check that kernel supports aes-xts-plain64 cipher")

ksponser, 2016-02-19:
Today I updated to kernel 4.1.18 (e6b537ef47d3d5c9d274fe35827ed4816f873520). After the reboot I can't open a luks container nor format a device with luks any more:

```
# cryptsetup luksOpen /dev/vg1/test1 luks-test1
Enter passphrase for /dev/vg1/test1:
Failed to setup dm-crypt key mapping for device /dev/vg1/test1.
Check that kernel supports aes-xts-plain64 cipher (check syslog for more info).
```

```
# cryptsetup luksFormat /dev/vg1/test2
WARNING!
========
This will overwrite data on /dev/vg1/test2 irrevocably.
Are you sure? (Type uppercase yes): YES
Enter passphrase:
Verify passphrase:
Failed to setup dm-crypt key mapping for device /dev/vg1/test2.
Check that kernel supports aes-xts-plain64 cipher (check syslog for more info). 
```

Even if the message tells to check syslog for more information, there's no entry in /var/log/messages nor in dmesg output.

Reverting to 4.1.17 (0764e7d78d30658f9bfbf40f8023ab95f952bcad) + reboot and everything is working again.

popcornmix, 2016-02-19:
This is what came from upstream with 4.1.18
https://cdn.kernel.org/pub/linux/kernel/v4.x/ChangeLog-4.1.18

Might be worth reading through these. There are a few crypto patches. Does anything look related to your use?

ksponser, 2016-02-19:
Thanks for the hint. The problem appeared in 4.1.18 upstream and I'm not the only one: [(https://bugzilla.kernel.org/show_bug.cgi?id=112631)]

popcornmix, 2016-02-19:
Okay, it's been reported to the right place. Post here if there is any new information.
Hopefully there will be a 4.1.19 along soon with a fix.

FuturePilot, 2016-02-21:
Was just about to report this. After going to 4.1.18 I couldn't open my LUKS disk anymore.

popcornmix, 2016-03-05:
I've pushed an update to 4.1.19 which looks to contain a fix for this. Please update and test.

richardski, 2016-03-06:
4.1.19 installed OK on RPi3 as per dmesg:

Linux version 4.1.19-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #851 SMP Sat Mar 5 22:58:42 GMT 2016
CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Machine model: Raspberry Pi 3 Model B Rev 1.2

Richard

popcornmix, 2016-03-08:
Closing after multiple reports that this is fixed.

closed 2016-03-08 by popcornmix

jkldgoefgkljefogeg, 2016-05-30:
device-mapper: reload ioctl on  failed: Invalid argument
Failed to setup dm-crypt key mapping for device /dev/sda1.
Check that kernel supports aes-cbc-plain cipher (check syslog for more info).

I am on 4.1.19, don't this this is fixed.

popcornmix, 2016-05-31:
4.1.19 is not the latest kernel. You should be on 4.4.
Are you using raspbian jessie? Are you up to date (``sudo apt-get update && sudo apt-get dist-upgrade``)

jkldgoefgkljefogeg, 2016-05-31:
Correct I am using kali-rolling (Debian testing).

popcornmix, 2016-05-31:
Okay, then that is not getting the supported kernel so we can't help with that.
I'd suggest you test with raspbian jessie and confirm if that is fixed.
If it is then you should report the issue to kali maintainers who should be able to update the kernel.

ptschack, 2017-06-26:
Unfortunately I'm having the same issue too, although I seem to be on kernel v4.9:
`root@box:/home/pi# uname -a`
`Linux box 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux`
I am up to date as per
`root@box:/home/pi# apt-get -y update && apt-get -y dist-upgrade`
When i try to run cryptsetup like this
`root@box:/home/pi# cryptsetup luksFormat -c aes-xts-plain64 -s 2048 -h sha512 -y /dev/sda1`
i get
`Einrichten der dm-crypt-Schlüsselzuordnung für Gerät »/dev/sda1« fehlgeschlagen.`
`Stellen Sie sicher, dass der Kernel die aes-xts-plain64-Verschlüsselung unterstützt.`
`(Sehen Sie im System-Log nach, ob sich dort Hinweise finden.)`
`Schlüsselgröße im XTS-Modus muss entweder 256 oder 512 Bits sein.`
Which is the same message as the OP, but in german. After some research I tried
`root@box:/home/pi# modprobe blkcipher`
`modprobe: FATAL: Module blkcipher not found.`

This is after a clean install of **2017-04-10-raspbian-jessie-lite.img**