Hexxeh/rpi-firmware #110: X unable to start with BRANCH=next

clebig, 2016-02-25:
I updated my firmware on a rpi2 today with BRANCH=next to move to the kernel 4.4.2-v7+ and since the upgrade, my Xorg doesn't want to start with my DMT monitor. However, when plugged on a TV, it works.
According to the xorg logs, X is unable to find initial modes, however I can see it detects the correct modes.
My xorg stack is up to date sticking on the raspbian jessie

This problem occurs too when using the last commit from the master branch.
I had to revert to commit f58f8848c236cf3222e67ccb587d38daaa8ae036 on master to get my system functional again.

Below some info.

Cheers.
[dpkg_xorg.txt](https://github.com/Hexxeh/rpi-firmware/files/147025/dpkg_xorg.txt)

Clément

`root@btv2-v02b:~# uname -a
Linux rbtv2-v02b 4.4.2-v7+ #844 SMP Thu Feb 18 14:24:58 GMT 2016 armv7l GNU/Linux
`

[tvservice_-m_dmt.txt](https://github.com/Hexxeh/rpi-firmware/files/147021/tvservice_-m_dmt.txt)
[dpkg_xorg.txt](https://github.com/Hexxeh/rpi-firmware/files/147020/dpkg_xorg.txt)
[config.txt](https://github.com/Hexxeh/rpi-firmware/files/147022/config.txt)
[edidparser_edid.dat_tvservice_-d.txt](https://github.com/Hexxeh/rpi-firmware/files/147023/edidparser_edid.dat_tvservice_-d.txt)

popcornmix, 2016-02-26:
I assume startx works if you remove ``dtoverlay=vc4-kms-v3d``?

clebig, 2016-03-09:
I confirme the problem persists with the last commit.

I also confirm startx works when I remove dtoverlay=vc4-kms-v3d

Clément

popcornmix, 2016-03-09:
Is the behaviour any different when you don't use BRANCH=next?
For general vc4-kms-v3d driver bugs you should report them here:
https://github.com/anholt/linux/issues

clebig, 2016-03-11:
Same behavior with BRANCH=next.

popcornmix, 2016-03-11:
Can you report this on anholt's github tracker. The display support is not handled by the GPU firmware but in his kernel source when vc4-kms-v3d is used.

closed 2016-03-11 by popcornmix