Hexxeh/rpi-firmware #111: RPF display touchscreen stops working

DougieLawson, 2016-03-13:
It looks like commit  2bd13b44c67f264e3a3545fcb938fd93ac35bb5c has broken the touch screen. I had to plug in a USB mouse to be able to move the cursor.

DougieLawson, 2016-03-13:
Reverting to 73eaed044ef7b58ecf81a8b164ea09c59fab9c7a got it back working on my RPi3B

popcornmix, 2016-03-13:
I don't have an LCD with me. I've tried reverting a possible cause - can you test:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_touch.zip

DougieLawson, 2016-03-13:
The rpi-update script doesn't handle *.dtbo files.

```
--- rpi-update  2016-03-13 18:07:43.084922170 +0000
+++ /usr/bin/rpi-update 2016-03-13 18:06:57.925047416 +0000
@@ -157,12 +157,12 @@
        cp "${FW_REPOLOCAL}/"*.dat "${FW_PATH}/"
        if [[ ${SKIP_KERNEL} -eq 0 ]]; then
                cp "${FW_REPOLOCAL}/"*.img "${FW_PATH}/"
-               if [[ -n $(shopt -s nullglob; echo "${FW_REPOLOCAL}/"*.dtb) ]]; then
-                       cp "${FW_REPOLOCAL}/"*.dtb "${FW_PATH}/"
+               if [[ -n $(shopt -s nullglob; echo "${FW_REPOLOCAL}/"*.dtb*) ]]; then
+                       cp "${FW_REPOLOCAL}/"*.dtb* "${FW_PATH}/"
                fi
-               if [[ -n $(shopt -s nullglob; echo "${FW_REPOLOCAL}/overlays/"*.dtb) ]]; then
+               if [[ -n $(shopt -s nullglob; echo "${FW_REPOLOCAL}/overlays/"*.dtb*) ]]; then
                        mkdir -p "${FW_PATH}/overlays"
-                       cp "${FW_REPOLOCAL}/overlays/"*.dtb "${FW_PATH}/overlays/"
+                       cp "${FW_REPOLOCAL}/overlays/"*.dtb* "${FW_PATH}/overlays/"
                        if [[ -f "${FW_REPOLOCAL}/overlays/README" ]]; then
                                cp "${FW_REPOLOCAL}/overlays/README" "${FW_PATH}/overlays/"
                        fi
```

popcornmix, 2016-03-13:
It does now...

closed 2016-03-13 by DougieLawson

popcornmix, 2016-03-13:
Can you confirm if dropbox firmware had working touchscreen?
If it does I'll push out an update.

popcornmix, 2016-03-13:
or did the updated rpi-update resolve the issue with the existing firmware?

DougieLawson, 2016-03-13:
The updated rpi-update (*.dtb changed to *.dtb*) fixed it.