Hexxeh/rpi-firmware #113: 1feae03ff210ab08c98e7bf59a2b17a215d4deb3 fails to init network

cribskip, 2016-03-31:
Hi,

just a quick note while I tested rbi-update with BRANCH=next. My Pi was updated to 1feae03ff210ab08c98e7bf59a2b17a215d4deb3 and boot failed to initialize the network device - at least.

I'm running headless and this was just quick look at next ;-)

Thank you very much for your efforts,
Sascha

popcornmix, 2016-03-31:
We've not had any other reports.
I think you'll need to attach a screen and see if there are any error messages visible.

Ruffio, 2016-07-06:
@cribskip do you have more informations?

cribskip, 2016-07-06:
I'm sorry but I don't have any more information and just wanted to point this out.
I'm now on master with 4.4.14-v7+ and very happy!

Thanks for your great work!

I think it's best to close as there's nothing more to it

closed 2016-07-06 by cribskip