Hexxeh/rpi-firmware #115: screen flicker after upgrade

mathlusiverse, 2016-04-21:
i just upgraded my Raspberry Pi 2.  Linux raspberrypi 4.4.7-v7+ #878
The screen now flicker and only trashcan appears. There is a white strip (about 1 cm tall) at the top that keep flashing. The trash can move up and down 1 cm higher as the white strip appears and disappear.

Please help.

popcornmix, 2016-04-21:
You updated with rpi-update? The screen is HDMI?

If you look at the commits here: https://github.com/Hexxeh/rpi-firmware/commits/master
each commit has a hash at the end of the URL

You can downgrade to a specific commit with:
``sudo rpi-update <hash>``

Can you try reverting to:
``sudo rpi-update 771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f``

which was the last 4.1 kernel, and report if that works?

mathlusiverse, 2016-04-21:
Hi @popcornmix:

>>>  You updated with rpi-update? The screen is HDMI?
Yes to both questions.

After reverting as suggested by your solution, the white strip is now occupied by the menu bar. The problem goes away, things back to normal, Thanks very much!

popcornmix, 2016-04-21:
That is not a complete solution. You may need to update in the future.
Can we confirm if the move to 4.4 was the start of the problem? Does
``sudo rpi-update c05b9e6fba9cc2ce9e47abf41b8d413b2a69d952``
work for you?

mathlusiverse, 2016-04-21:
The problem started when I run 'sudo rpi-update' yesterday.  I don't have access to my Raspberry Pi 2 now. I'll try that later.

mathlusiverse, 2016-04-21:
I just tried 'sudo rpi-update c05b9e6fba9cc2ce9e47abf41b8d413b2a69d952'.
It does not work, I get the same appearing/disappearing of the menu bar (white strip at the top).
I have to revert to 'sudo rpi-update 771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f' again.

popcornmix, 2016-04-21:
Can you get a photo (or video recoding) of the screen? I'm not quite understanding what you are seeing.
No one else is reporting an issue, so can you explain anything unique about your setup?
Are you running raspbian jessie? What sort of display is it? Any other peripherals?
Any custom config.txt settings? Any GUI related packages installed?

pelwell, 2016-04-21:
@popcornmix I think what @mathlusiverse is seeing is the menu bar appearing briefly as a white bar,  shifting other screen content (just the trashcan) down, then disappearing and moving the trashcan up again. In other words LXPanel is crashing and restarting in a loop.

@spl237 suggests editing ~/.config/lxpanel/LXDE-pi/panels/panel and deleting all the plugin entries. Then try restarting the GUI and adding the plugins one at a time.

mathlusiverse, 2016-04-21:
I don't have any custom custum config.txt, I don't have raspbian jessie. I have opencv, gstreamer and may be something else that i forgot. But I haven't install any new software between my last working rpi-update and yesterday.

I did not make any changes to ~/.config/lxpanel/LXDE-pi/panels/panel.  I will try another update, and play with the panel file and see what happen.

mathlusiverse, 2016-04-21:
Attached is the file  ~/.config/lxpanel/LXDE-pi/panels/panel. 

`# lxpanel <profile> config file. Manually editing is not recommended.
# Use preference dialog in lxpanel to adjust config when you can.

Global {
  edge=top
  allign=left
  margin=0
  widthtype=percent
  width=100
  height=36
  transparent=0
  tintcolor=#000000
  alpha=0
  autohide=0
  heightwhenhidden=2
  setdocktype=1
  setpartialstrut=1
  usefontcolor=0
  fontsize=12
  fontcolor=#ffffff
  usefontsize=0
  background=0
  backgroundfile=/usr/share/lxpanel/images/background.png
  iconsize=36
}
Plugin {
  type=menu
  Config {
    name=Menu
    image=/usr/share/raspberrypi-artwork/raspitr.png
    system {
    }
    separator {
    }
    item {
      name=Run...
      image=gnome-run
      command=run
    }
    separator {
    }
    item {
      name=Shutdown...
      image=gnome-logout
      command=logout
    }
  }
}
Plugin {
  type=space
  Config {
    Size=8
  }
}
Plugin {
  type=launchbar
  Config {
    Button {
      id=/usr/share/applications/epiphany-browser.desktop
    }
    Button {
      id=/usr/share/applications/pcmanfm.desktop
    }
    Button {
      id=/usr/share/applications/lxterminal.desktop
    }
    Button {
      id=/usr/share/applications/wolfram-mathematica.desktop
    }
    Button {
      id=/usr/share/applications/wolfram-language.desktop
    }
  }
}
Plugin {
  type=space
  Config {
    Size=8
  }
}
Plugin {
  type=taskbar
  expand=1
  Config {
    tooltips=1
    IconsOnly=0
    ShowAllDesks=0
    UseMouseWheel=1
    UseUrgencyHint=1
    FlatButton=0
    MaxTaskWidth=200
    spacing=1
    GroupedTasks=0
  }
}
Plugin {
  type=space
  Config {
    Size=2
  }
}
Plugin {
  type=tray
  Config {
  }
}
Plugin {
  type=dhcpcdui
  Config {
  }
}
Plugin {
  type=space
  Config {
    Size=2
  }
}
Plugin {
  type=volumealsa
  Config {
  }
}
Plugin {
  type=space
  Config {
    Size=2
  }
}
Plugin {
  type=cpu
  Config {
    ShowPercent=1
    Foreground=#a9a9a9a9a9a9
    Background=#d3d3d3d3d3d3
  }
}
Plugin {
  type=dclock
  Config {
    ClockFmt=%R
    TooltipFmt=%A %x
    BoldFont=0
    IconOnly=0
    CenterText=0
  }
}
Plugin {
  type=space
  Config {
    Size=2
  }
}
`

mathlusiverse, 2016-04-21:
I re-update the Pi and removed all plugin entries in  ~/.config/lxpanel/LXDE-pi/panels/panel   and reboot, the screen still flicker. The menu bar is appearing/disappearing.

spl237, 2016-04-21:
If you can get a terminal window open, try entering "sudo pkill lxpanel"; that should stop the flicker and leave you with a screen with no menu bar. Then restart lxpanel from the terminal by entering "lxpanel -p LXDE-pi" and see what messages come up - if you post the result here, I can see if there is anything that looks suspicious.

spl237, 2016-04-21:
Also you say above that you are running Wheezy and not Jessie. I'm not sure if the latest kernel is compatible with Wheezy, which is now over 6 months out of date.

@popcornmix - has anyone tested the latest kernel with Wheezy? Is it possible there is an incompatibility with init.d or something like that?

mathlusiverse, 2016-04-21:
@spl237:  the flicker goes away after "sudo pkill lxpanel", attached is the message of "lxpanel -p LXDE-pi".

BTW, is there a way to confirm if I have  Wheezy or Jessie?

```
** Message: x-terminal-emulator has very limited support, consider choose another terminal

** (lxpanel:3795): WARNING **: The directory '~/Templates' doesn't exist, ignoring it

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-can-change-accels after class was initialised

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popup-delay after class was initialised

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(lxpanel:3795): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-button-images after class was initialised

(lxpanel:3795): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed

(lxpanel:3795): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed

(lxpanel:3795): Gtk-CRITICAL **: IA__gtk_misc_set_alignment: assertion 'GTK_IS_MISC (misc)' failed
** Message: Connecting ...
** Message: Status changed to down

(lxpanel:3795): Gtk-CRITICAL **: IA__gtk_dialog_response: assertion 'GTK_IS_DIALOG (dialog)' failed
** Message: Status changed to opened
** Message: Connected to dhcpcd-6.7.1
** Message: Status changed to connected
** Message: wlan0: Associated with DSU_Wireless
** Message: wlan0: Configured 192.168.117.250/22
** Message: eth0: Link is down
** Message: wlan0: Received scan results

** (lxpanel:3795): WARNING **: volumealsa: Default ALSA device not valid - resetting to internal
Segmentation fault
pi@raspberrypi ~ $ 
```

spl237, 2016-04-21:
It looks to me from the panels file you posted above that you are definitely still running Wheezy - the Jessie update added a plugin called ejecter, which you don't have.

The log above suggests that the volumealsa plugin is causing the crash - if you remove that from the panels file, by deleting the lines:

Plugin {
  type=volumealsa
  Config {
  }
}

and then try and start it again with lxpanel -p LXDE-pi - does the segfault still happen?

popcornmix, 2016-04-21:
The newer device tree kernel probably has the snd-bcm2835 disabled. Maybe the volumealsa panel doesn't like that?
Does adding ``dtparam=audio=on`` to config.txt help?

mathlusiverse, 2016-04-21:
Thanks!  I removed the volumealsa plugin and reboot, everything is back to normal now!
It seems the problem is solved.

I'll update to Jessie later.

mathlusiverse, 2016-04-21:
I didn't do anything to the config.txt yet, so it seems that I don't need
      dtparam=audio=on

spl237, 2016-04-21:
@popcornmix Yes, the volumealsa panel uses the onboard sound device as an ultimate fallback - it expects that it will always be able to find it. If it can't, bad things are likely to happen. If we are changing setup so that the onboard audio can be disabled or enabled with devicetree, I might need to revisit the plugin and give it some alternative default behaviour.

mathlusiverse, 2016-04-21:
Thanks everyone for maintaining Raspberry Pi and solving the problems so efficiently!

PrackBraska, 2016-05-06:
I just spent a few hours starting and stopping lxpanel and disabling plugins and then I found this. I was able to get mine working by disabling volumealsa as well. What can I do to avoid/fix this going forward?

spl237, 2016-05-06:
Make sure you have the latest version of lxpanel installed - at minimum, you need to be running Jessie; robustness against this problem has been improved on several occasions. If you are already running the latest version, wait a week or two - there is another version due for release soon which contains further improvements.

PrackBraska, 2016-05-06:
Looks like I have the latest version of lxpanel installed. I'd like to upgrade to Jessie but I'm not able to do a clean install at this time. I'm running a web forum on one of my pi's and I'm not sure how difficult it's going to be to migrate that website off and back on to the clean install. Thank you for the information.

richhaynes, 2016-07-04:
Firstly, thank you for all the comments on this issue. Removing the 'volumealsa' section from my lxpanel config temporaily solved the flickering although I have lost my volume button :(

Secondly, I would like to add something that I believe may be related. After running rpi-update I noticed I now recieve fails when I shutdown that are also related to ALSA. These errors are below:

[FAIL] Shuting down ALSA... warning: 'alsactl store' failed with error message 'alsactl: save_state:1580: No soundcards found...' ... failed
[FAIL] startpar: service(s) returned failure: alsa-utils ... failed

Both these errors started after the update from 4.1 to 4.4. I am also using wheezy.

I hope this extra information can help!

Extra info:
cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
cat /proc/version
Linux version 4.4.14-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #896 SMP Sat Jul 2 15:09:43 BST 2016

popcornmix, 2016-07-04:
Have you tried adding ``dtparam=audio=on`` to config.txt?

richhaynes, 2016-07-05:
Hi popcornmix,

Thanks for pointing that out. I somehow missed that in the conversation!

I have added the line and my ALSA errors are gone. I have also readded the volumealsa plugin and everything works just fine.

dtparam=audio=on is the correct fix for this issue :)

closed 2016-07-05 by popcornmix

paulwratt, 2016-09-18:
I got this too, lucky I have an ancient desktop which has lxterm and a folder on it, and I dont boot to X, otherwise (....)

I added the line to config, but in fact was surprised not to see in rem'd out.

I thought that  dtparam=audio=on (or some other audio setting) has been in use for some time, or was that only in the Jessie image?

Anyway, a post install update check and fix would solve any further Wheezy problems for future users.

BTW with this problem, using raspi-config to set audio also has issues

(I'm about to reboot - wish me luck)

mstechly, 2017-05-03:
For me changing audio didn't help.
After downgrading to 4.1.20, issue stopped. (I've chosen this version randomly.)

fyoemg, 2017-06-12:
@ popcornmix  

# issue. 'after rpi-update,  menu bar flickering '
Do follow bellow:
sudo rpi-update 771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f

which was the last 4.1 kernel, and report if that works?  

It's working ! Thank @popcornmix