Hexxeh/rpi-firmware #116: Current Firmware 4.4.7 breaks sonic-pi

soslug, 2016-04-21:
Looking for confirmation so reporting an issue with latest firmware update and sonic-pi and audio specifically.

Problem repeated on B+, RPi2 and RPi3

Further info sudo apt-get update and sudo apt-get upgrade run on pre builds no issues on rpi-update and after reboot sonic-pi does not run. Possible audio problem rather than sonic-pi issue. Appears related to rpi-update.

Steps :-
1: sudo apt-get update
2: sudo apt-get upgrade
3: sonic-pi runs -no issues
4: sudo reboot
5: after many reboots for some reason after startx started sonic-pi runs ok no issues
6: close sonic-pi
7: sudo rpi-update 
8: ran sonic-pi no issues
9: sudo reboot
10: run startx many reboots reason unknown
11: eventually login run startx sonic-pi fails to run error reports jackd not loaded

popcornmix, 2016-04-21:
We are aware of a problem with Sonic Pi and the 4.4 kernel.
The issue is that jackd fails to init with multichannel support in alsa driver (https://github.com/raspberrypi/linux/commit/688a5f0daa45e0a51b324707768d472e1d715c13).
We suspect that adding ``-i 2 -o 2`` to the jackd command may avoid the problem, but haven't got confirmation from the Sonic Pi author.

The issue will be resolved one way or the other before next firmware/kernel apt update.

moor-terces, 2016-04-27:
I just updated to firmware 4.4.8 and it is still an issue. Even after manually starting jack before opening Sonic-Pi the audio quality is jacked ;D. The audio sounds like it is bitcrushed. I have tried several packages that use jack and although other programs are able to start jack, the audio quality issue remains.

popcornmix, 2016-04-27:
Can you edit ``/opt/sonic-pi/app/server/sonicpi/lib/sonicpi/scsynthexternal.rb``
Find the line:
``sys("jackd -R -T -p 32 -d alsa -n 3 -p 2048 -r 44100& ")``
Edit to:
``sys("jackd -R -T -p 32 -d alsa -n 3 -p 2048 -r 44100  -i2 -o2& ")``
Does that help?

soslug, 2016-04-28:
Having proceeded with a full update and upgrade of raspbian software plus the installation of 4.4.v7 firmware along with the changes to the above file I can confirm having tried it twice that the patch does indeed work. A reboot was required to initiate the change after editing the file, however I have not tried any other application other than Sonic-pi that uses the audio output so treat my analogy of the situation with caution.

popcornmix, 2016-04-28:
The next update to SonicPi inclues "-i 2 -o 2" so that should be taken care of.
I'll close this, but if you spot any other incompatibilities can you create issues here:
https://github.com/raspberrypi/linux/issues
(this repo is purely for distribution of firmware binaries - the linux repo will have the right devs watching).

closed 2016-04-28 by popcornmix

moor-terces, 2016-04-28:
This fix did work for Sonic-Pi. Thank you for the fix!

I am still having issues with other programs. Can you give a quick explanation of what the additonal code is doing so I can dig into the other packages and try and reproduce the fix there?

Thank you.

popcornmix, 2016-04-29:
The 4.4 kernel supports multichannel audio over alsa.
jackd by default tries to use the maximum number of channels, so that will be 8 with newer kernel and 2 with older kernel. That doesn't seem to work. The ``-i 2 -o 2`` limits the input and output channels to 2 which allows jackd to start up and so lets sonic pi run.