Hexxeh/rpi-firmware #117: problem with wlan0 using 4.4.8

dh-connect, 2016-04-30:
since I've update raspberry pi 2 from 4.4.7 to 4.4.8 with usb wlan (Edimax EW-7811Un 150Mbps 11n Wi-Fi USB Adapter) wlan doesn't connect automatically with booting  

syslog:
wpa_supplicant[1716]: nl80211: Driver does not support authentication/association or connect commands

but when i run ifup wlan0, it connects, do you have an idea what the error means?

popcornmix, 2016-05-01:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

dh-connect, 2016-05-08:
i went back to 4.4.7 still the same issue, so i guess it's wasn't the firmware update

dh-connect, 2016-05-08:
in iwconfig wlan0 has changed it's name to enx74da386baca7, i don't know how this has happend, do you know how to name it back? i think that's why wlan0 can't connect

popcornmix, 2016-05-08:
Sorry I don't know. The forum is probably a better place for questions like that.

dh-connect, 2016-05-08:
I get this here in syslog: rtl8192cu 1-1.2:1.0 enx74da386baca7: renamed from wlan0
so the error can be driver related

XECDesign, 2016-05-08:
@dh-connect what does your `/lib/udev/rules.d/75-persistent-net-generator.rules` look like?

dh-connect, 2016-05-08:
by the way also eth0 has been renamed, i have another raspberrypi 1, same thing eth0 has been renamed,
both pi's use debian 9 strech, there was also an update of wireless-tools, i don't know if this caused the problem

here the content of /lib/udev/rules.d/75-persistent-net-generator.rules

```
# These rules generate rules to keep network interface names unchanged
# across reboots and write them to /etc/udev/rules.d/70-persistent-net.rules.

# variables used to communicate:
#   MATCHADDR           MAC address used for the match
#   MATCHID             bus_id used for the match
#   MATCHDRV            driver name used for the match
#   MATCHIFTYPE         interface type match
#   COMMENT             comment to add to the generated rule
#   INTERFACE_NAME      requested name supplied by external tool
#   INTERFACE_NEW       new interface name returned by rule writer

ACTION!="add",                          GOTO="persistent_net_generator_end"
SUBSYSTEM!="net",                       GOTO="persistent_net_generator_end"

# ignore the interface if a name has already been set
NAME=="?*",                             GOTO="persistent_net_generator_end"

# new predictable network interface naming scheme
# http://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/
IMPORT{cmdline}="net.ifnames"
ENV{net.ifnames}=="1",                  GOTO="persistent_net_generator_end"

# device name whitelist
KERNEL!="ath*|msh*|ra*|sta*|ctc*|lcs*|hsi*", \
                                        GOTO="persistent_net_generator_end"

# ignore Xen virtual interfaces
SUBSYSTEMS=="xen",                      GOTO="persistent_net_generator_end"

# ignore UML virtual interfaces
DRIVERS=="uml-netdev",                  GOTO="persistent_net_generator_end"

# ignore "secondary" raw interfaces of the madwifi driver
KERNEL=="ath*", ATTRS{type}=="802",     GOTO="persistent_net_generator_end"

# ignore "secondary" monitor interfaces of mac80211 drivers
KERNEL=="wlan*", ATTRS{type}=="803",    GOTO="persistent_net_generator_end"

# by default match on the MAC address and interface type
ENV{MATCHADDR}="$attr{address}"
ENV{MATCHIFTYPE}="$attr{type}"

# match interface dev_id
ATTR{dev_id}=="?*", ENV{MATCHDEVID}="$attr{dev_id}"

# These vendors are known to violate the local MAC address assignment scheme
# Interlan, DEC (UNIBUS or QBUS), Apollo, Cisco, Racal-Datacom
ENV{MATCHADDR}=="02:07:01:*", GOTO="globally_administered_whitelist"
# 3Com
ENV{MATCHADDR}=="02:60:60:*", GOTO="globally_administered_whitelist"
# 3Com IBM PC; Imagen; Valid; Cisco; Apple
ENV{MATCHADDR}=="02:60:8c:*", GOTO="globally_administered_whitelist"
# Intel
ENV{MATCHADDR}=="02:a0:c9:*", GOTO="globally_administered_whitelist"
# Olivetti
ENV{MATCHADDR}=="02:aa:3c:*", GOTO="globally_administered_whitelist"
# CMC Masscomp; Silicon Graphics; Prime EXL
ENV{MATCHADDR}=="02:cf:1f:*", GOTO="globally_administered_whitelist"
# Prominet Corporation Gigabit Ethernet Switch
ENV{MATCHADDR}=="02:e0:3b:*", GOTO="globally_administered_whitelist"
# BTI (Bus-Tech, Inc.) IBM Mainframes
ENV{MATCHADDR}=="02:e6:d3:*", GOTO="globally_administered_whitelist"
# Novell 2000
[... 51 lines cut ...]
 ENV{COMMENT}="PCI device $attr{vendor}:$attr{device}"
SUBSYSTEMS=="pcmcia", \
 ENV{COMMENT}="PCMCIA device $attr{card_id}:$attr{manf_id}"
SUBSYSTEMS=="usb", \
 ENV{COMMENT}="USB device 0x$attr{idVendor}:0x$attr{idProduct}"
SUBSYSTEMS=="ccwgroup", \
 ENV{COMMENT}="S/390 device at $id"
SUBSYSTEMS=="ibmveth", \
 ENV{COMMENT}="LPAR virtual device at $id"
SUBSYSTEMS=="ieee1394", \
 ENV{COMMENT}="Firewire device $attr{host_id}"
ENV{COMMENT}=="", \
 ENV{COMMENT}="Unknown $env{SUBSYSTEM} device ($env{DEVPATH})"
ATTRS{driver}=="?*", \
 ENV{COMMENT}="$env{COMMENT} ($attr{driver})"

# ignore interfaces without a driver link like bridges and VLANs, otherwise
# generate and write the rule
DRIVERS=="?*", IMPORT{program}="write_net_rules"

# rename the interface if requested
ENV{INTERFACE_NEW}=="?*", NAME="$env{INTERFACE_NEW}"

LABEL="persistent_net_generator_end"
```

XECDesign, 2016-05-08:
That looks okay. Is this raspbian, ubuntu or something else?

dh-connect, 2016-05-08:
raspbian on debian base

XECDesign, 2016-05-08:
Actually, I think that file just ensures that  if you plug in a second wifi dongle and it shows up as wlan1, it will show up as wlan1 each time, so that wouldn't have anything to do with it.

A bit more info on actual predictable interface names:
https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/

I am not entirely sure what you mean by 'raspbian on debian base'. What does `apt-cache policy` say?

dh-connect, 2016-05-08:
it says
```
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://debian.yacy.net ./ Packages
     release c=
     origin debian.yacy.net
 500 http://archive.raspberrypi.org/debian jessie/ui armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=jessie,l=Raspberry Pi Foundation,c=ui,b=armhf
     origin archive.raspberrypi.org
 500 http://archive.raspberrypi.org/debian jessie/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=jessie,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
 500 http://download.owncloud.org/download/repositories/stable/Debian_8.0  Packages
     release o=obs://obs.int.owncloud.com/ce:9.0/Debian_8.0,n=Debian_8.0,l=ce:9.0,c=
     origin download.owncloud.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/rpi armhf Packages
     release o=Raspbian,a=testing,n=stretch,l=Raspbian,c=rpi,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/non-free armhf Packages
     release o=Raspbian,a=testing,n=stretch,l=Raspbian,c=non-free,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/contrib armhf Packages
     release o=Raspbian,a=testing,n=stretch,l=Raspbian,c=contrib,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
     release o=Raspbian,a=testing,n=stretch,l=Raspbian,c=main,b=armhf
     origin mirrordirector.raspbian.org
```

XECDesign, 2016-05-08:
Given that you're on stretch, that's most likely the intended behaviour. Try adding `net.ifnames=1` to the end of `cmdline.txt`, keeping everything in one line, if you would like to disable it.

dh-connect, 2016-05-08:
didn't work

XECDesign, 2016-05-08:
Probably best to take it to the forum, since it's not a firmware issue.

Ruffio, 2016-07-06:
@dh-connect have your issue been resolved? If that is the case, then please close this issue. Thanks.

Ruffio, 2016-07-26:
@dh-connect anything new on this?

dh-connect, 2016-07-29:
yes can be closed, after an update of debian stretch network interface was renamed back to wlan0 and it worked, I'm on 4.4.13 now

closed 2016-07-29 by popcornmix