Hexxeh/rpi-firmware #118: Problem with sound output on Raspberry Pi 2 & 3 under Ubuntu 16.04

DMChatterton, 2016-05-03:
Since updating to the latest rpi firmware, there was no response to the Test Speakers buttons in the Sound panel, neither were there any clicks from the speakers when shutting down. All sound appeared to be muted. I tested this using different Raspberry Pi boards and HDMI monitors (with speakers) and the problem appeared to be due to accepting the update dated 30 April (kernel:snd-bcm2835 etc.)
I performed an rpi-update 384626a4e08d49262fd0ebb799b82f330ca7a0df using the commit code from the previous update (dated 27 April) and my sound system returned to normal.
There is still a problem with the sound system in that the Test Speakers responses (Front Left and Front Right) are broken and incomplete but it is possible for the system to reproduce continuous unbroken sound when playing YouTube videos via Firefox.
If I am to continue to try out other firmware updates as they become available, I would appreciate advice as to how to blacklist the rogue update otherwise I am stuck with the update on 27 April.

pelwell, 2016-05-03:
Add `dtparam=audio=on` to /boot/config.txt and reboot.

DMChatterton, 2016-05-03:
Thanks for the advice but that statement is already present in 
/boot/config.txt under the section Base Device Tree Parameters at the 
end of the file.

David

popcornmix, 2016-05-04:
From latest firmware can you report output from:
```
vcgencmd version
uname -a
sudo vcdbg log msg
aplay -l
```
and post your config.txt

DMChatterton, 2016-05-04:
The output from the specified terminal commands is copied into the 
attached text file.

The content of the config.txt file is attached.

Regards,

David

david@david-desktop:~$ vcgencmd version
May  3 2016 17:44:01 
Copyright (c) 2012 Broadcom
version b870a5bbf6c662327916ee582c22309c9f997b3d (clean) (release)
david@david-desktop:~$ uname -a
Linux david-desktop 4.4.8-v7+ #881 SMP Sat Apr 30 12:16:50 BST 2016 armv7l armv7l armv7l GNU/Linux
david@david-desktop:~$ sudo vcdbg log msg
[sudo] password for david: 
002651.946: HDMI:EDID version 1.3, 1 extensions, screen size 60x34 cm
002652.010: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr444; sRGB is default colourspace; preferred format is native; does not support GTF
002652.131: HDMI:EDID found unknown detail timing format: 2560x1440p hfp:48 hs:32 hbp:80 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
002652.170: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
002652.207: HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
002652.246: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
002652.283: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
002652.323: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
002652.362: HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
002652.401: HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
002652.488: HDMI:EDID standard timings block x 8: 0x8180 81C0 8100 A9C0 B300 D1C0 0101 0101 
002652.541: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
002652.594: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
002652.645: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
002652.698: HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 3
002652.750: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 4
002652.801: HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 5
002665.804: HDMI:EDID parsing v3 CEA extension 0
002665.843: HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
002665.890: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
002665.934: HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
002665.975: HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
002666.016: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
002666.054: HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
002666.088: HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
002666.120: HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
002666.155: HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
002666.188: HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
002666.220: HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz 
002666.253: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
002666.287: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
002666.320: HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
002666.352: HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz 
002666.386: HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz 
002666.419: HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
002666.452: HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz 
002666.486: HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz 
002666.517: HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
002666.571: HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
002666.590: HDMI:EDID found HDMI VSDB length 5
002666.616: HDMI:EDID HDMI VSDB has physical address 1.0.0.0
002666.633: HDMI:EDID HDMI VSDB has no extension fields
002666.676: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
002666.896: HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
002688.154: HDMI:Setting property pixel encoding to Default
002688.174: HDMI:Setting property pixel clock type to PAL
002688.193: HDMI:Setting property content type flag to No data
002688.212: HDMI:Setting property fuzzy format match to enabled
002888.237: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
002900.370: hdmi: HDMI:EDID version 1.3, 1 extensions, screen size 60x34 cm
002900.439: hdmi: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr444; sRGB is default colourspace; preferred format is native; does not support GTF
002900.567: hdmi: HDMI:EDID failed to find a matching detail format for 2560x1440p hfp:48 hs:32 hbp:80 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
002900.589: hdmi: HDMI:EDID calculated refresh rate is 60 Hz
[... 1247 lines cut ...]
##
##     Default mmc.
##
#dtparam=pwr_led_trigger=mmc

## pwr_led_activelow
##     Set to "on" to invert the sense of the LED
##
##     Not available on Model A/B boards.
##
##     Default off.
##
#dtparam=pwr_led_activelow=off

## pwr_led_gpio
##     Set which GPIO to use for the PWR LED
##
##     In case you want to connect it to an external device
##
##     Not available on Model A/B boards.
##
##     Default 35.
##
#dtparam=pwr_led_gpio=35
##

popcornmix, 2016-05-06:
I've pushed a commit to linux tree (and rpi-update kernel is rebuilt). Can you test?

DMChatterton, 2016-05-06:
I have executed: sudo rpi-update from the terminal and the sound seems to be restored (after a reboot).
Uname -r is now reporting 4.4.9-v7+

There is still an issue with the quality of the Front Left and Front Right responses to the sound test buttons but the lack of sound seems to have been solved.

Thanks you for your efforts.

closed 2016-05-06 by DMChatterton

DMChatterton, 2016-05-07:
I have also tested the playback of sound under Rasbian Jesse after updating to 4.4.9-v7+.
Using Iceweasel to select a YouTube video, the sound seems to play back via the HDMI output to a monitor at the expected quality.

LanceJZ, 2017-08-24:
I just installed and updated Ubuntu Mate at 1.16.2 on my Pi 3B. I have no sound. I checked, and the confit.txt is the same as Daves. I need sound, and I want it to come out of the 1/8th audio jack.

pelwell, 2017-08-24:
What value does `amixer cget numid=3` report? 0 means auto-detect, 1 means analogue jack and 2 means HDMI. You can force jack output using `amixer cset numid=3 1`.

LanceJZ, 2017-08-24:
It came back numid=3 I'll change that to 1.

pelwell, 2017-08-24:
It's confusing, but numid is the number of the control - it's the  "values = " output that matters.

LanceJZ, 2017-08-24:
Yeah, I still hear nothing. Ouput 0.00db, Dummy Output.

pelwell, 2017-08-24:
What does `alsamixer` show? If you type `alsamixer > am.txt` then hit Ctrl-C you should get a screen capture in am.txt.

LanceJZ, 2017-08-24:
Sorry, I gave up. I'm going back to Raspian. Ubuntu is not ready for prime time in Pi.

Tropicalrambler, 2018-01-23:
@pelwell Your post solved it for me basically.  I typed alsamixer and got this screen with the obvious problem.
[image]

As you can see the dB gain is set to -102.38 by default.  With `alsamixer`open, you can just press the up ▲ arrow on the keyboard and volume comes up.  In my particular case, I was testing the setup of an [AirPlay receiver using shairport](https://pimylifeup.com/raspberry-pi-airplay-receiver/) and I had music already being piped to the pi.  Cranking the volume up yielded results.  I suggest you crank it up to 0.00dB - Item: PCM [dB gain: 0.00]

Come to think of it, I had some issues updating my children's PIPER computer earlier this year, where the same problem ensued. That community is not as savvy as the usual linux user, thus they were having serious issues with their updates, which updated the firmware and caused this problem on their modified Raspbian OS.

I am enjoying some music right now, as soon as I am done, I will reboot and confirm if it came back.

Tropicalrambler, 2018-01-23:
OK.. Found this: [Fix alsa configuraiton for future reboots](https://unix.stackexchange.com/questions/210113/default-sound-volume-for-all-alsa-devices)

When you exit alsamixer, run this command:

`alsactl store`