Hexxeh/rpi-firmware #119: No sound after updating firmware today.

joswart, 2016-05-07:
RPI 2.

# lsmod | grep snd
snd_bcm2835            23163  0
snd_pcm                95441  1 snd_bcm2835
snd_seq                58506  0
snd_seq_device          5266  1 snd_seq
snd_timer              22396  2 snd_pcm,snd_seq
snd                    68368  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device

# aplay -l
aplay: device_list:252: no soundcards found...

Needed to downgrade to a 4.1.x kernel to get it working again:
sudo rpi-update dea25fa62132365c11087e51e416df656db28bf3

popcornmix, 2016-05-07:
Do you have:
dtparam=audio=on
In config.txt?

joswart, 2016-05-07:
Nope I didn't. Adding it and updating again fixed it.

Must have missed that change. Thanks for helping out.

XECDesign, 2016-05-07:
@joswart If you're running raspbian, What does `apt-cache policy raspberrypi-sys-mods` return? That line should have been added automatically a while back.

joswart, 2016-05-07:
$ sudo apt-cache policy raspberrypi-sys-mods
N: Unable to locate package raspberrypi-sys-mods

XECDesign, 2016-05-07:
and just `apt-cache policy`? I am guessing this isn't an install from the raspberrypi.org image or that it has been stripped down?

joswart, 2016-05-07:
$ sudo apt-cache policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.org/debian/ wheezy/main armhf Packages
     release o=Raspberry Pi Foundation,a=oldstable,n=wheezy,l=Raspberry Pi Foundation,c=main
     origin archive.raspberrypi.org
 500 http://raspberrypi.collabora.com/ wheezy/rpi armhf Packages
     release o=Collabora,n=wheezy,l=Collabora Raspberry Pi graphics enablement,c=rpi
     origin raspberrypi.collabora.com
 500 http://mirror.netcologne.de/raspbian/raspbian/ wheezy/rpi armhf Packages
     release v=7.0,o=Raspbian,a=oldstable,n=wheezy,l=Raspbian,c=rpi
     origin mirror.netcologne.de
 500 http://mirror.netcologne.de/raspbian/raspbian/ wheezy/non-free armhf Packages
     release v=7.0,o=Raspbian,a=oldstable,n=wheezy,l=Raspbian,c=non-free
     origin mirror.netcologne.de
 500 http://mirror.netcologne.de/raspbian/raspbian/ wheezy/contrib armhf Packages
     release v=7.0,o=Raspbian,a=oldstable,n=wheezy,l=Raspbian,c=contrib
     origin mirror.netcologne.de
 500 http://mirror.netcologne.de/raspbian/raspbian/ wheezy/main armhf Packages
     release v=7.0,o=Raspbian,a=oldstable,n=wheezy,l=Raspbian,c=main
     origin mirror.netcologne.de
Pinned packages:

I really can't remember what I used to install this system, but since it was my first raspberry, so I guess I went fairly default. In other words, I would be surprised if I didn't get the image from raspberrypi.org.

XECDesign, 2016-05-07:
Ah... you're still on wheezy, which is no longer supported, so that makes sense now.

DMChatterton, 2016-05-07:
Please see my last message in [issue 118](https://github.com/Hexxeh/rpi-firmware/issues/118).
To test the new kernel and firmware under Jessie, I restored a disk image from the Rasbian Jessie image: 2016-03-18-raspbian-jessie.img downloaded from the RPi website using Ubuntu's Disks app. After booting Jessie on the Raspberry Pi 3, I executed rpi-update to bring the kernel and firmware to the latest version (4.4.9-v7+) as well as updating the cache and upgrading any software via apt-get. I then installed Iceweasel and used it to play a YouTube video which it did using the speakers on my HDMI monitor. I suffered no problems with the sound quality at all. Hence I closed the issue as having been solved.

Ruffio, 2016-07-06:
@joswart as I read your response, this issue has been solved. Please close this issue if it has been resolved. Thanks.

closed 2016-07-06 by joswart