Hexxeh/rpi-firmware #121: SPI writes not working after 4.4.x updates

msedv, 2016-06-08:
After rpi-upgrade'ing 8 raspis all of the SPI hardware+applications used by us (mainly based on SPI-Py, device_tree disabled) stopped working:

* no error messages in the ioctl-calls!
* reads work
* writes are simply ignored (!)

Testing (many...) different firmware versions with different raspbian versions and different SPI-applications (own and foreign) I got the following "result":

* 4.1.21 is the last working version
* 4.4.6 and 4.4.7 crashes my test pi (a B+ model) - rainbow screen
* 4.4.8 lets the pi boot and everything works fine - except the SPI writes
* same for all later versions including 4.4.13 from today

Any ideas? Thank you!
Markus

volschin, 2016-06-08:
Did you read this?
https://www.raspberrypi.org/documentation/configuration/device-tree.md

pelwell, 2016-06-08:
If reads are working and everything worked on 4.1.21 I seriously doubt it's a Device Tree problem.

volschin, 2016-06-08:
Chapter 4.4:

> Since the switch to the 4.4 kernel and the use of more upstream drivers, Device Tree usage is required in RPi kernels.

pelwell, 2016-06-08:
...except for

> (mainly based on SPI-Py, device_tree disabled)

Trying to use 4.4 with DT disabled is not supported and unlikely to work. For that reason, trying to disable DT on an official 4.4 kernel is ignored.

Ruffio, 2016-07-06:
@msedv have you solved your issue or can you provide some more information? logs? error messages?

pelwell, 2016-07-06:
I believe SPI-Py is broken and used to only work by accident.  See here for details and a fix for the library: https://github.com/raspberrypi/linux/issues/1547

msedv, 2016-07-06:
@Ruffio Sorry, no - it simply doesn't give results, no error messages, no logs. BUT:
@pelwell This (simple) patch seems the solution - it looks like there was an issue with the SPI-Py module which didn't occur with older kernel version - but with the latter ones. I think I'll have to dig a little bit deeper into the SPI speficiation and reactivate the datalogger...

pelwell, 2016-07-11:
Since the 4.4 SPI driver is working as it should - the fault is in the SPI-Py module (I've raised an issue with the author [here](https://github.com/lthiery/SPI-Py/issues/17)) - can we close this?

msedv, 2016-07-11:
@pelwell IMHO yes.

closed 2016-07-13 by popcornmix