Hexxeh/rpi-firmware #122: lvl0: Error creating SDL window!

nschloe, 2016-06-18:
Commit a16f516e5d058eaa9a2d22288576c58f24db6137 introduces errors of the kind
```
lvl0: Error creating SDL window!
Could not initialize OpenGL / GLES library
lvl0: Renderer failed to initialize!
lvl0: Window failed to initialize!
```
See [here](https://retropie.org.uk/forum/topic/2028/error-creating-sdl-window) for more info.

As a workaround, one can revert back with
```
sudo rpi-update a16f22120d7979dc91e8d2e03fecb611fb0259e0
```

loganmarchione, 2016-06-18:
@nschloe I confirmed that fixes my issue, thanks!

otlet, 2016-06-18:
I had this same problem. Problem exist and no "fix"

popcornmix, 2016-06-18:
Can you confirm if it is the kernel or firmware (start.elf`/fix up.dat) that introduced the problem?

ED6E0F17, 2016-06-19:
Looking at the commit, the symlink for "opt/vc/lib/libGLESv1_CM.so" got changed.

popcornmix, 2016-06-19:
Does ``sudo ldconfig`` help?

ED6E0F17, 2016-06-19:
@popcornmix -  "libGLESv2.so" is a symlink you created for "libGLESv2.so" ; it is not created by the Userland build.

https://www.raspberrypi.org/forums/viewtopic.php?t=10075&p=168652

popcornmix, 2016-06-20:
Can you test again after ``rpi-update``?

popcornmix, 2016-06-21:
@nschloe okay to close?

nschloe, 2016-06-22:
Tested and confirmed working.

closed 2016-06-22 by nschloe