Hexxeh/rpi-firmware #123: Latest firmware upgrade incompatible with GQRX under UbuntuMATE

DMChatterton, 2016-07-03:
The Software Defined Radio app has run acceptably on the RPi 3 under Ubuntu MATE (16.04).
However, having applied the firmware revision offered on 2 July, GQRX refused to run.
I reverted back to the firmware upgrade offered on 27 June and GQRX was able to run as before.
One of the firmware upgrades offered on 2 July (I don't know which at this stage) is incompatible with GQRX.

David

popcornmix, 2016-07-03:
Can you continue here: https://github.com/raspberrypi/firmware/issues/625
as this sounds like the same issue.

closed 2016-07-03 by popcornmix

DMChatterton, 2016-07-03:
I am not sure that this is the same issue at that to which you refer. GQRX does not run Rasbian so there is no way of knowing whether the most recent firmware upgrade has affected it. I would prefer to keep this open in order to attract comments from those also using GQRX under Ubuntu MATE.
This issue has been reported to the GQRX forum and those interested might well be looking in this direction.

popcornmix, 2016-07-03:
Does https://github.com/raspberrypi/firmware/issues/625#issuecomment-230163519 make it work again?

DMChatterton, 2016-07-03:
> Does raspberrypi/firmware#625 (comment)
> make it work again?

I got GQRX working again by going back to the RPi firmware update 
released on 27 Jun. That works for me on the RPi 3 which is running 
GQRX. If I have to avoid future rpi-updates to keep this working, then 
so be it. If anything more complex is required to be done to get GQRX to 
run under the latest firmware, I am hoping that the providers of GQRX 
will come up with a simple solution.

I have several RPi 2s that I have upgraded with the latest firmware. 
Since these are not running GQRX, there have not been any issues as yet 
although I have not tested them exhaustively with all the apps that I 
have installed.