Hexxeh/rpi-firmware #124: Problem with usb microphone

kenken64, 2016-07-23:
[ 4044.146384] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:0D8C:013C.0040/input/input63
[ 4044.200036] hid-generic 0003:0D8C:013C.0040: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.2/input2
[ 4044.230041] usb 1-1.2: USB disconnect, device number 76
[ 4044.599459] usb 1-1.2: new full-speed USB device number 77 using dwc_otg
[ 4044.699855] usb 1-1.2: device descriptor read/all, error -32
[ 4044.779418] usb 1-1.2: new full-speed USB device number 78 using dwc_otg
[ 4044.886128] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=013c
[ 4044.886146] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4044.886160] usb 1-1.2: Product: USB PnP Sound Device
[ 4044.886172] usb 1-1.2: Manufacturer: C-Media Electronics Inc.      
[ 4044.887290] snd-usb-audio 1-1.2:1.0: cannot find the slot for index 0 (range 0-0), error: -16
[ 4044.887308] usb 1-1.2: cannot create card instance 0
[ 4044.887334] snd-usb-audio: probe of 1-1.2:1.0 failed with error -16
[ 4044.892424] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:0D8C:013C.0041/input/input64
[ 4044.950234] hid-generic 0003:0D8C:013C.0041: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.2/input2
[ 4048.750152] usb 1-1.2: USB disconnect, device number 78

kenken64, 2016-07-23:
pi@acorn2:~ $ lsusb
Bus 001 Device 012: ID 0bda:0151 Realtek Semiconductor Corp. Mass Storage Device (Multicard Reader)
Bus 001 Device 079: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 010: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 009: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Ruffio, 2017-01-10:
@kenken64 do you still experience this issue? Are you sure it is a kernel bug?

Ruffio, 2017-01-13:
@kenken64 do you still experience this issue?

Ruffio, 2017-04-05:
@popcornmix I don't believe that this is the correct repo for this kind of issue. This issue should be closed due to lack of response for more than 3 months...

closed 2017-04-05 by popcornmix