Hexxeh/rpi-firmware #125: Webcam not working using this branch. Ok with stable

qdel, 2016-08-09:
Hello,

Up-to-date raspbian, using
```
$> sudo BRANCH=next rpi-update
 *** If no errors appeared, your firmware was successfully updated to 47d64831ccbc99080f98c1bc8948717c3416cb0f
```

after reboot:
```
$> raspistill -v

raspistill Camera App v1.3.8

Width 3280, Height 2464, quality 85, filename (null)
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled
```

Doing only:
```
$> sudo rpi-update
 *** If no errors appeared, your firmware was successfully updated to e7178c06ac5b527085ceb380aa56f1ed52558ea0
```

after reboot:
```
raspistill -v

raspistill Camera App v1.3.8

Width 3280, Height 2464, quality 85, filename (null)
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Closing down
Close down completed, all components disconnected, disabled and destroyed
```

Camera is enabled, split is 256.

popcornmix, 2016-08-10:
what does ``vcgencmd get_camera`` report?
And ``vcgencmd version``?

What were you on before updating to BRANCH=next?
It would be useful if you can confirm if master branch of rpi-update works.
``sudo rpi-update`` should return you to bleeding edge of master branch.

qdel, 2016-08-10:
Using stable:

```
$> vcgencmd get_camera 
supported=1 detected=1
$> vcgencmd version
Jul 28 2016 12:46:32 
Copyright (c) 2012 Broadcom
version e12916091ba9d68ef2780e4e142ade56aa301754 (clean) (release)
```

Using next:
```
$> vcgencmd get_camera 
supported=0 detected=0
$> vcgencmd version
Aug  8 2016 11:47:45 
Copyright (c) 2012 Broadcom
version b86cd0444798d376c8c7e955dea5dabf22f4abe4 (clean) (release)
```

Yes, rpi-update work perfectly, i can switch to next or stable without any kind of problem (i make test on the same pi, switching with rpi-update and reboot just after it.)

popcornmix, 2016-08-10:
Seems to be a bug in the updated bootcode.bin where it doesn't load start_x.elf.
If you were to rename ``start_x.elf`` to ``start.elf`` and ``fixup_x.dat`` to ``fixup.dat`` that should provide a temporary solution.

Ping @ghollingworth for the bootcode fix.

qdel, 2016-08-11:
Using this fix, is it still possible to boot directly on usb device?

popcornmix, 2016-08-11:
Should be fine

popcornmix, 2016-08-15:
``BRANCH=next`` has been updated with a fix for this. Renaming the start files should no longer be necessary. Please test it is okay.