Hexxeh/rpi-firmware #126: The latest update screwed the whole network on RPi3

foobar13373, 2016-08-11:
After loading the latest firmware (https://github.com/Hexxeh/rpi-firmware/commit/8c1304684a4f8a353254a987903bb7f7bb8f6324) the cable-network on my Pi3 totally stopped working ( the integrated WLAN seemed to work, but I don't have a WLAN so I could not test actually). I had to roll back to the previous version which made it work instantly again (rollback means in this case: inserting the SD card into my Ubuntu laptop and copy the previous revision `.img` files to `/boot`). I do not own an older model, so I can not tell if the problem is Pi3 specific.

I can not tell what exactly was wrong. With the new firmware no interface came up automatically (including loopback). The usual commands `ifup`, `ifdown`, `ifconfig` and so on acted totally normal however. Even `dmesg `didn't give any error or warning. The devices where fully recognized (`ifconfig -a` listed them with their correct names as before), but the network could not establish any link.

Even the loopback interface didn't come up at first. I could however get it to work manually with `sudo ifconfig lo up`. Pinging it (`ping localhost`) worked then again. On the Ethernet interface I was unlucky. It came up without logged errors, but it did not had any address assigned (it is configured to be static).

Trying to also configure the WLAN in `/etc/network/interfaces` seemed to work, I could bring it up and it had the static address assigned which I gave it. But since I do not have a WLAN I can not test it really (if a wireless link could actually be established).

I'm on Raspbian stretch (the past few days or even weeks didn't have any significant update to anything network-related).

pelwell, 2016-08-11:
I am running the same firmware version without any problems on a Pi3 with wired ethernet.

Shortly after booting, can you run `dmesg > dmesg.txt` and upload the file somewhere - pastebin, a gist, or even as a comment (please use three backticks before and after the log to disable markup processing in the log)?

foobar13373, 2016-08-17:
It seems the new firmware was indeed not the problem. On the same day a `hostapd` update was released which fro some reason put some files into `/etc/network/interfaces/{post,pre}-up.d/`, which were false or non-existant (I didn't check). This stopped the networks coming up. I still wonder why localhost worked manually and wlan didn't throw any error and why restoring the old firmware "solved" it.

Nevertheless, with the current firmware and purged hostapd it now works again.

closed 2016-08-17 by foobar13373