Hexxeh/rpi-firmware #127: Dropped RX packets - RPi 1, 2, and 3

stamster, 2016-09-08:
Hello,
I have multiple RPi's running on two locations. I noticed that each of them report dropped packets, but only in RX direction.

RPi 3:

```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:xx:xx:xx  
          inet addr:192.168.100.20  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::906e:f3:xxxx:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:275242 errors:0 dropped:15336 overruns:0 frame:0
          TX packets:71301 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:38165404 (36.3 MiB)  TX bytes:23844353 (22.7 MiB)
```

RPi 2:
Just rebooted after latest rpi-firmware patch, and only 4 mins of uptime getting: `RX packets:689 errors:0 dropped:5 overruns:0 frame:0`

RPi 1 model B:
`RX packets:647 errors:0 dropped:28 overruns:0 frame:0`

RPi 1:
`RX packets:209844 errors:0 dropped:14892 overruns:0 frame:0`

The only single RPi which does not seem to be affected by this issue is another RPi 1 model B, where I didn't upgraded firmware.
So if it's a firmware related issue, the last known good firmware/kernel w/o RX drops is this one:

```
4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux

Mar 15 2016 14:48:20 
Copyright (c) 2012 Broadcom
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 (clean) (release)

10 days uptime, and not a single drop:

```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:xx:xx:xx  
          inet addr:192.168.100.30  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1335268 errors:0 dropped:0 overruns:0 frame:0
          TX packets:721742 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:81649608 (77.8 MiB)  TX bytes:185224562 (176.6 MiB)
```

RPi's behave exactly the same on other networks, so it's not LAN related, since other PC's and devices do not have any drops with uptime of 250 days.

stamster, 2016-09-10:
Am I the only one experiencing this issue? :8ball:

MrEngman, 2016-09-10:
No you're not. It's a fairly common issue, although the number of dropped packets can vary considerably.

stamster, 2016-09-11:
Thanks for response @MrEngman 
So for now, the only solution would be to revert to proven firmware I guess.

stamster, 2016-09-25:
I have upgraded again to the new fw release, but drops are still there.

stamster, 2016-09-26:
Additionally, I installed OSMC for what it is made for :) And guess what - there are no drops on primary eth interface. Running Debian/GNU Linux Jessie 8.5. I'm not sure what firmware they're using?

Ruffio, 2017-04-05:
@stamster I don't believe that this is the correct repo for this kind of issue. This Git is only a placeholder for the firmware files. You should use this Git instead: https://github.com/raspberrypi/firmware/issues/

This issue should be closed.

popcornmix, 2017-04-05:
Almost certainly a kernel (https://github.com/raspberrypi/linux/issues/) rather than a firmware issue (https://github.com/raspberrypi/firmware/issues/) so that would be a more suitable place.

closed 2017-04-05 by popcornmix