Hexxeh/rpi-firmware #128: Missing shared library :  libelftoolchain.so

bousqi, 2016-09-18:
I just upgraded my raspberry pi 3 with rpi-update tool to the last 4.4.21 kernel.

`Linux rpi-stable 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux`

Now, I can't run the following command due to a missing dependency
>   $ sudo vcdbg
   vcdbg: error while loading shared libraries: libelftoolchain.so: cannot open shared object file: No such file or directory

By the way if I check deps on the vcdbg binary, this specific lib is not found

>      $ ldd vcdbg
        linux-vdso.so.1 (0x7eed2000)
        /usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f4d000)
        libdebug_sym.so => /opt/vc/lib/libdebug_sym.so (0x76f29000)
        libelftoolchain.so => not found
        libvcos.so => /opt/vc/lib/libvcos.so (0x76f0f000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ee7000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76ed4000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76ebc000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76de0000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76d65000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76d38000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76bf7000)
        /lib/ld-linux-armhf.so.3 (0x54b5a000)

Any idea on how to solve this ?

bousqi, 2016-09-18:
I did had an issue while upgrading to 4.4.21. 
Having a service on my system that keep tvservice running, the first run of rpi-update failed.
So I stopped it and restart rpi-update. Does the second run succeed but I lead to this problem

popcornmix, 2016-09-18:
@bousqi you are correct. Can you try updating again?

bousqi, 2016-09-18:
Perfect. 
Issue can be closed.

closed 2016-09-18 by bousqi