Hexxeh/rpi-firmware #129: Green LED remains On when Pi shutdown

DragonflyNet67, 2016-10-06:
I Updated my Pi2B to Kernel 4.4.23 (#913) and after Shutdown or Poweroff the **green LED remains on**. This did not happen with Kernel 4.4.22 (#912). On SSH Shutdowns the Green LED is important to see, if the Pi is shutdown and not writing to the sd-card anymore. As far as I know, the green LED was the only indicator to tell if the Pi could be safely unplugged.

Ruffio, 2017-01-10:
@DragonflyNet67 I have just downloaded and installed the latest official image from the download page and don't experience this issue. Do you still experience this issue? Could it be something that are written to the disk during shutdown that just takes a long time? Since there are no one else that have reported this issue, it look like you are the only one with this issue, so it could be specific to your setup.

Ruffio, 2017-01-13:
@DragonflyNet67 Do you still experience this issue?

Ruffio, 2017-01-22:
@popcornmix please consider to close this. I can reproduce the issue and @DragonflyNet67  hasn't responded to two times ping...

closed 2017-02-20 by popcornmix