Hexxeh/rpi-firmware #131: audit support not in kernel

nambrosch, 2016-12-05:
i'm running fedora 25 with kernel/modules/firmware provided via rpi-update.  upon boot i'm having issues loading the systemd-journald service:

```
# systemctl --failed
  UNIT                            LOAD   ACTIVE SUB    DESCRIPTION
● auditd.service                  loaded failed failed Security Auditing Service
● systemd-journald.service        loaded failed failed Journal Service
● systemd-logind.service          loaded failed failed Login Service
● systemd-journald-dev-log.socket loaded failed failed Journal Socket (/dev/log)
● systemd-journald.socket         loaded failed failed Journal Socket
```

i believe it's because there is no support for audit in the provided kernel.  here's the output of running it manually:

```
# /sbin/auditd -f
Config file /etc/audit/auditd.conf opened for parsing
local_events_parser called with: yes
write_logs_parser called with: yes
log_file_parser called with: /var/log/audit/audit.log
log_group_parser called with: root
log_format_parser called with: RAW
flush_parser called with: INCREMENTAL_ASYNC
freq_parser called with: 50
max_log_size_parser called with: 8
num_logs_parser called with: 5
priority_boost_parser called with: 4
qos_parser called with: lossy
dispatch_parser called with: /sbin/audispd
name_format_parser called with: NONE
max_log_size_action_parser called with: ROTATE
space_left_parser called with: 75
space_action_parser called with: SYSLOG
action_mail_acct_parser called with: root
admin_space_left_parser called with: 50
admin_space_left_action_parser called with: SUSPEND
disk_full_action_parser called with: SUSPEND
disk_error_action_parser called with: SUSPEND
use_libwrap_parser called with: yes
tcp_listen_queue_parser called with: 5
tcp_max_per_addr_parser called with: 1
tcp_client_max_idle_parser called with: 0
enable_krb5_parser called with: no
krb5_principal_parser called with: auditd
distribute_network_parser called with: no
Error - audit support not in kernel
Cannot open netlink audit socket
The audit daemon is exiting.
```

can you confirm?  if this is the case, is it possible to add support going forward?

```
[root@raspberrypi ~]# uname -a
Linux raspberrypi 4.4.36-v7+ #933 SMP Fri Dec 2 22:03:01 GMT 2016 armv7l armv7l armv7l GNU/Linux
```

thanks.

popcornmix, 2016-12-05:
rpi-update is for raspbian. See https://github.com/Hexxeh/rpi-update

If using a distribution with a custom kernel you shouldn't use rpi-update unless the maintainer of the distribution recommends it.

nambrosch, 2016-12-05:
thanks for the quick response.  the instructions i followed included rpi-update.  this was a while ago (fedora 23, i've since updated to 24 and 25) so don't recall which they were.

i will try a vanilla installation at some point and look to see if there is another way to install firmware without rpi-update.

in the mean time do you have any thoughts on audit support?  i'm curious if it's excluded on purpose.

popcornmix, 2016-12-05:
We are concerned about performance and memory costs of config options like these.
Before enabling we'd like strong evidence that it doesn't harm these. See:
https://lwn.net/Articles/600568/

popcornmix, 2016-12-05:
Note you can run:
```sudo SKIP_KERNEL=1 rpi-update```
to get latest gpu firmware without updating the kernel.

nambrosch, 2016-12-05:
i agree, auditd isn't something i really want however fedora's journald depends on auditd (see below) on all of their architectures:

```
systemd-journald.service:After=systemd-journald.socket systemd-journald-dev-log.socket systemd-journald-audit.socket syslog.socket
systemd-journald.service:Sockets=systemd-journald.socket systemd-journald-dev-log.socket systemd-journald-audit.socket
```

i'm reading over their rpi page it looks like they've put some work specific to rpi into fedora 25 which they had not previously, so that could be the saving grace here.

https://fedoraproject.org/wiki/Raspberry_Pi#When_will_support_for_Fedora_24_or_23_arrive.3F

nambrosch, 2016-12-06:
update - a vanilla fedora 25 installation contains a kernel with all the appropriate modules to satisfy dependencies for their services.  thanks for sending me in the right direction.

closed 2016-12-06 by nambrosch