Hexxeh/rpi-firmware #133: cec strange behaviour after last firmware update

hashashin, 2016-12-31:
Hi, after the last firmware update I'm experiencing problems with my TV (samsung). It's powering on himself randomly. Also a pair of cec-client alias I've to power on/off the TV via ssh stop working too. Sometimes I'm capable of powering off the TV, but then never comes on again via command.

alias tvon="echo 'on 0' | cec-client -s"
alias tvoff="echo 'standby 0' | cec-client -s"

If I revert to af9cb14d5053f89857225bd18d1df59a089c171e all works fine again.

Thanks, salud.

PS: sorry about my poor english skills.

popcornmix, 2017-01-02:
Do you have any cec related settings in config.txt?

hashashin, 2017-01-02:
This is what I've there (removed comments):

disable_overscan=1
arm_freq=1300
over_voltage=2
core_freq=500
sdram_freq=483
over_voltage_sdram_p=0
over_voltage_sdram_i=0
over_voltage_sdram_c=0
gpu_mem_1024=256
overscan_scale=1
dtparam=i2c_arm=on
dtparam=sd_overclock=100
dtparam=audio=on
gpu_mem_256=128
gpu_mem_512=256

popcornmix, 2017-01-04:
Can you update and report if your issue is gone?

hashashin, 2017-01-04:
Ok, seems like the tv now is not randomly power on himself and the cec-client commands are working too.
But If I use those commands I lose control over kodi via the Tv remoter, not a drama, but annoying, as that was working before. I guest is good as is, I suppose the change in cec that you don't revert is mean to fix another things... so.

Thanks, salud.

popcornmix, 2017-01-04:
> But If I use those commands I lose control over kodi via the Tv remoter, not a drama, but annoying, as that was working before.

Are you sure about that? I wouldn't expect that behaviour to change.
My understanding was that libcec was one client only, so when kodi was running you couldn't use cec-client.

hashashin, 2017-01-04:
Gonna try that with af9cb14d5053f89857225bd18d1df59a089c171e again to be sure.

hashashin, 2017-01-04:
Yeah, you're right, my bad. So all good now, thanks again.

Salud.

popcornmix, 2017-01-04:
The cec change has only been temporarily reverted.
My colleague will investigate next week and assuming he finds a problem, there will be another firmware to test. I'll ping you here when that is ready.

hashashin, 2017-01-04:
👍

hashashin, 2017-01-15:
Hi, seems like 68f0062a1491706150b9165670d9f842c7362972 reintroduced that behavior (random power on), it's now less frequent but it's there. Now also if you are watching the tv it will change to kodi randomly (I can't remember that last time, or at least nobody in the house complains about this before). That only happens on plain TV channels if I use another cec ready hardware in another hdmi port, ps3 in this case, the tv remains there.

Thanks, salud.

popcornmix, 2017-01-15:
Can you test this firmware?
https://drive.google.com/uc?id=0B-6zmEDJwxZEd3BTYVR1Ynl4Rlk&export=download
It includes CEC cleanup patches 1-5 which should help to narrow down the issue.

hashashin, 2017-01-15:
Done, lets past some time now, I'll post results here in about an hour? to be safe.

popcornmix, 2017-01-15:
Yes - make sure you are confident whether it is good or bad.

hashashin, 2017-01-15:
Ok, this one seems good, no power on, no input change from tv to hdmi.

popcornmix, 2017-01-15:
Can you test this firmware?
https://drive.google.com/uc?id=0B-6zmEDJwxZEV0Jvc0d0VEJodWM&export=download
It includes CEC cleanup patches 1-8.

hashashin, 2017-01-15:
Just powered on itself right now.

hashashin, 2017-01-15:
Strange, after that 1st power on, I rebooted again without touching anything, power off the tv when full booted to kodi and now still remains powered off...

popcornmix, 2017-01-15:
Assuming you are reporting last firmware was bad, then:
https://drive.google.com/uc?id=0B-6zmEDJwxZESTZZTFNBRjVUQ1E&export=download
It includes CEC cleanup patches 1-7.

hashashin, 2017-01-16:
I think the 1-8 one it's fine, after all the day running seems it's not powering on the tv.

popcornmix, 2017-01-16:
https://drive.google.com/uc?id=0B-6zmEDJwxZEb2RxS3J5UFFmcWM&export=download
includes CEC cleanup patches 1-9

hashashin, 2017-01-16:
Patches 1-9 almost 5h running, no random power on tv.

popcornmix, 2017-01-16:
There is very little in cleanup 10, so result is a little surprising, but for completeness can you test:
https://drive.google.com/uc?id=0B-6zmEDJwxZEaWV2ZEdLSHBJbzA&export=download
includes CEC cleanup patches 1-10

hashashin, 2017-01-16:
Now I'm surprised too, no power on for one hour with the last one(1-10). Then I put back master and seems to be good for now also (another hour). Perhaps was just my tv acting weird this time, I dunno. 
I'll make noise here if this comes back and I'm absolutely sure is pi fault.

Ruffio, 2017-01-22:
@hashashin so this is fixed?

popcornmix, 2017-01-23:
@hashashin I'll assume the thumbs up response means you are not currently experiencing the CEC issue with latest firmware. I'll close this for now.
Feel free to reopen if you are confident a specific firmware is working less well than a previous version.

closed 2017-01-23 by popcornmix