Hexxeh/rpi-firmware #134: 4.9.2 won't boot on my 3B with an RPF display and everything on a USB stick

DougieLawson, 2017-01-12:
It hangs at the rainbow screen.

I've reverted to bcc6146e102d85b1aa214855ad7aae278d3bd269 and everything is back to normal.

popcornmix, 2017-01-12:
This is using boot from USB (no sdcard)?
Did it break in the latest update (i.e. did https://github.com/Hexxeh/rpi-firmware/commit/af04ebcdf1dfa2d979487ed35fb037a99f1552bc work?)

DougieLawson, 2017-01-12:
This is with just a Sandisk 32GB USB stick. I booted it with a 2017-01-11 Raspbian SDCard chrooted it and ran rpi-update to get the USB stick back to the known good level it worked OK.

I'll try https://github.com/Hexxeh/rpi-firmware/commit/af04ebcdf1dfa2d979487ed35fb037a99f1552bc for you.

DougieLawson, 2017-01-12:
https://github.com/Hexxeh/rpi-firmware/commit/af04ebcdf1dfa2d979487ed35fb037a99f1552bc works OK.

popcornmix, 2017-01-12:
Could you test latest master branch? That should have usb/net boot support now.
It will probably behave the same, but would be interesting if it worked.

asavah, 2017-01-12:
No, master from this repo will not work.
Both USB and PXE end up in ranbow screen.

Log excerpt from my tftp server
```
Jan 12 16:45:07 nas2 in.tftpd[24941]: RRQ from 192.168.36.181 filename bootcode.bin
Jan 12 16:45:07 nas2 in.tftpd[24945]: RRQ from 192.168.36.181 filename config.txt
Jan 12 16:45:07 nas2 in.tftpd[24947]: RRQ from 192.168.36.181 filename start.elf
Jan 12 16:45:09 nas2 in.tftpd[24948]: RRQ from 192.168.36.181 filename fixup.dat
```
After that network link on pi goes down and I get a rainbow and blinking act led (blink 7? times, pause, repeat).

DougieLawson, 2017-01-12:
latest master branch 8f43681f762e4f632b31572f31f42d405ba9a663 hangs in exactly the same way as the head of the next branch.

PeterOGB, 2017-01-12:
Same problem as asavah.  Just upgraded to "next" and it's stoppped flashing green led 7 times.

popcornmix, 2017-01-12:
Pushed a fix. Both master and next branches should now be supporting USB/Net boot.

asavah, 2017-01-12:
Works like a charm now.
Thx.

DougieLawson, 2017-01-12:
c8f7c2c62702aa16334608343a8cb2a409538b8b 4.4.41-v7+ works
548450f1d2fd29f57d8855a6076284478f280b41 4.9.2-v7+ works 

Fixed in under 24 hours, that's absolutely fantastic. Thank you so much.

closed 2017-01-13 by popcornmix