Hexxeh/rpi-firmware #135: GPU core is halting

jcjordyn130, 2017-01-26:
every time I try to use kodi I get a black screen, when I check the GPU logs I get this:
vchiq_core: header f6ca35d8 (msgid 0) - size 4002 too big for slot
vcos_about: Halting

I've tried on kernels 4.4.42, 4.4.44, and 4.8.x. I've also tried with the overclock turned off.

Ruffio, 2017-04-05:
@jcjordyn130 is this still an issue?

Ruffio, 2017-04-05:
@jcjordyn130 You should use this Git instead: https://github.com/raspberrypi/firmware/issues/
This repo is only a placeholder for files.

popcornmix, 2017-04-05:
Seeing as there are no problems with kodi on raspbian, osmc, libreelec etc it does suggest the problem may lie with how Ubuntu Mate has built kodi, the kernel config options used, or the firmware and/or opt/vc libs included as such may be better to report it to the Ubuntu Mate maintainers.

If it seems to be a firmware regression (i.e. it was working until a firmware update broke it, then open an issue at the firmware repo).

closed 2017-04-05 by popcornmix