Hexxeh/rpi-firmware #137: Power consumption after shutdown extremely high

vssystemluba, 2017-02-14:
After shutting down my rpi 3 with `sudo shutdown -h now` or `sudo poweroff`, I used to see a quiescent power draw of less than 100mA by the pi.  However, I happened to look at the power draw more recently and running a firmware update and on the same hardware (also with no USB peripherals), the power draw of the rpi after a software shutdown is now much higher, around 220mA.  

We are using a small portable battery pack to power a pi in a robotics project, and we need the internal battery to stop supplying power to the pi once it has shut down.  I'm fairly confident that a firmware update has caused the increased power draw, but I can't isolate which update would've done this.  Can you take a look and see if I'm on the right track here?

Thanks!

popcornmix, 2017-02-15:
Is this 4.4 or 4.9 kernel?
Can you try ``sudo halt``?

vssystemluba, 2017-02-15:
Kernel looks like 4.4.20.  I notice the same behavior when I update to the latest via `rpi-update`.

After a `sudo halt`, power consumption is the same as `sudo shutdown -h now`, roughly 220mA.

popcornmix, 2017-02-15:
``sudo halt`` should output "Reached target shutdown" to console and then flash ACT LED ten times. Are you seeing that?

vssystemluba, 2017-02-15:
I don't have screen access to the actual pi inside the robot.  I always SSH to access it.  I therefore can't directly test the console output after the `sudo halt` command on that pi.

However, I did set up another brand new pi 3 with the same SD card from the robot and booted the pi up and eventually issued the `sudo halt` command.  The console shows `[ ok ] Reached target shutdown` as expected, and the ACT LED flashes 10 times.  I also noticed that on the brand new pi outside of the robot, the power consumption after shutting down via `sudo halt`, `sudo shutdown -h now`, or `sudo poweroff` is reasonable, always around 60mA.

This is also interesting, however.  Does this mean that the USB devices or GPIO pins continue supplying power after the pi has shut down?

popcornmix, 2017-02-15:
You'll need to check the LEDS to see if the pi actually has shutdown.

halt/shutdown is a polite shutdown request.
It will wait for any systemd services to shutdown.
My guess is you have something that is hanging and we never reach shutdown.
If the shutdown code is reached you will see the ten ACT LED flashes.

vssystemluba, 2017-02-15:
But would I expect to see the two different behaviors on two different pis with the same SD card running exactly the same thing?  The only difference between the pi inside the robot and the one outside that's definitely shutting down properly are attached USB devices and GPIO pins.

vssystemluba, 2017-02-15:
I was just able to confirm that for the pi in the robot, when I send a `sudo halt` the ACT LED flashes 10 times and then goes off.  After this, the power LED is on solid, the ACT LED is off, and the power draw of the pi in the robot is a steady 220mA.

popcornmix, 2017-02-15:
Can you test each pi3, with no peripherals attached and booting from the same sdcard.
(i.e. everything identical except the pi3 you are using)
Does one have high current after halt and one has low?

vssystemluba, 2017-02-15:
This is not easily doable as I would need to find a technician to uninstall the pi from the robot.  The uninstall would take 4-6 hours and finding a technician with availability would take a few days.  The pi in the robot was a brand new pi 3 which we got from the same source as the one I'm now testing externally.

I was able to get a schematic of the robot and confirmed that we are powering a number of peripherals from the 5V and 3.3V pins on the pi, as well as having a number of peripherals attached to the GPIO pins.  Would the 5V and 3.3V pins continue supplying power once the pi is shut down?

popcornmix, 2017-02-15:
Yes, power supplies will continue to run after shutdown (as you can see from the power LED).
So possibly something you have attached to the Pi is consuming the unaccounted for 160mA.

vssystemluba, 2017-02-15:
Okay.  I didn't realize that the 5V pins and 3.3V pins on the pi are directly attached to the power rails and aren't controllable via software.  I've confirmed that this is indeed where the extra power draw is coming from.

I may implement a transistor on a GPIO pin to control the power output from the pi.  This would allow me to "shut off" the 5V and 3.3V power pins to the peripherals and therefore reduce power draw of the "pi".

I'll close this issue since I don't think this is resolvable via firmware.  Thank you so much for your timely responses and helpful information!

closed 2017-02-15 by vssystemluba

pelwell, 2017-02-15:
It might be worth searching the forums for gpio-poweroff - a module and overlay that can be used, with external hardware, to really power off the Pi.