Hexxeh/rpi-firmware #143: Where's the script ?

flatmax, 2017-02-28:
I am after the script you use to copy the binaries from your kernel directory to the rpi-firmware directory before pushing here.

In other words :
If I build the pi kernel in /build/linux
Can you give the script to copy the necessary files in the required layout from /build/linux to /build/rpi-firmware

thanks
Matt

lurch, 2019-06-09:
I'm not quite sure what you're asking for - does https://www.raspberrypi.org/documentation/linux/kernel/building.md help?

lurch, 2019-06-09:
And I've just added https://github.com/Hexxeh/rpi-firmware/wiki if that helps?

stappersg, 2019-12-10:
And who has the power to close this issue?
@flatmax you?    (and sorry for the late response)

Ruffio, 2020-08-26:
@flatmax please close this issue :-)

closed 2020-08-26 by flatmax