Hexxeh/rpi-firmware #145: Raspberry 1 model B does not boot with latest boot code.bin

rgc2000, 2017-04-02:


pelwell, 2017-04-03:
On the Pi that fails to boot, but using the good bootcode.bin, what does `grep Revision /proc/cpuinfo` return?

rgc2000, 2017-04-03:
Revision is 000e but I don't think this will help. Both working and non working Pi have the same cpuinfo 
  processor      : 0
  model name     : ARMv6-compatible processor rev 7 (v6l)
  BogoMIPS       : 697.95
  Features       : half thumb fastmult vfp edsp java tls
  CPU implementer        : 0x41
  CPU architecture: 7
  CPU variant    : 0x0
  CPU part       : 0xb76
  CPU revision   : 7

  Hardware       : BCM2835
  Revision       : 000e
  Serial         : 00000000xxxxxxxx

pelwell, 2017-04-03:
Can you try with [these firmware files](https://drive.google.com/file/d/0B_P-i4u-SLBXWDRKTFp1MzR3MkE/view?usp=sharing)? Functionally they should be the same as pre-issue firmwares, but after booting if you run `sudo vcdbg log msg |& grep DEBUG` it will show some register values that will help to debug this problem.

rgc2000, 2017-04-03:
I've installed your firmware. Here is DEBUG output : 
```
000693.990: DEBUG RSTS=00001020
000694.106: DEBUG RSTS=00001020
000694.140: DEBUG RSTS=00001020
000837.764: DEBUG SPARE=00000000
000837.777: DEBUG RSTS=00001020
000843.378: DEBUG SPARE=00000000
000843.389: DEBUG RSTS=00001020
000877.603: DEBUG SPARE=00000000
000877.615: DEBUG RSTS=00001020
```

pelwell, 2017-04-03:
Thanks - that eliminates a few avenues of investigation.

pelwell, 2017-04-03:
Can you try reverting to 41561448388d90f19f2e40ac62ce1c21c2e064ec, but with the new (bad) bootcode.bin ?

rgc2000, 2017-04-03:
Ok, reverted to 41561448388d90f19f2e40ac62ce1c21c2e064ec kernel is 4.9.17+
With provided bootcode.bin : boot OK
with "bad" bootcode.bin : boot OK too
Interesting. Looks like bootcode alone is not the only faulty part

pelwell, 2017-04-03:
That's interesting, but not entirely surprising.

The change(s) that triggered the failure makes bootcode.bin store the number of the boot partition in a register called PM_SPARE, for start.elf to read. The initial bug reports were because, due to an undocumented hardware feature, one of the bits in that register was already being read by start.elf and being used to disable initialisation of the SMPS. SInce this "other" reader had been there for a long time, reverting bootcode.bin was the only workaround. The fix, or so we thought, was to avoid that specific bit, and initial reports suggested that was successful in preventing the boot failures.

The second wave of boot failures (I'm aware of this issue and  https://github.com/raspberrypi/firmware/commit/e5ca26f0395d7aa62f6af77c1371f8a9f65edf20#commitcomment-21608499) seems to be different. The first affected all Pis using BCM2708C0, whereas the second seems to be limited to certain instances. I had wondered whether something in some Pis was causing the signficant bits in PM_SPARE to be written unexpectedly, causing the wrong boot partition to be selected, but the debug trace you acquired for me shows that it is zero.

gnuheidix, 2017-04-04:
There are also other people affected and rolling back worked for them as well. see: https://archlinuxarm.org/forum/viewtopic.php?f=31&t=11540

pelwell, 2017-04-04:
Here's [another test firmware](https://drive.google.com/file/d/0B_P-i4u-SLBXN1JwVTlEMWVwSG8/view?usp=sharing) - `start*.elf`s and `fixup*.dat`s. `rpi-update` to the current head of the master branch (9495159f5bd1bfeccb89a7e2166fb40837517fe5) then, before rebooting (unless you are happy to update the card by other means, since your Pi won't boot), copy the contents of the .zip into /boot. 

After the reboot, check the firmware log for debug messages:
```
pi@raspberrypi:~$ sudo vcdbg log msg |& grep SPARE
001092.828: SD: SPARE=400000, part=0
001153.320: PF: SPARE=400000, part=0
001207.119: PF: SPARE=400000, part=0
```
I hope you see something different, otherwise I'll be clueless.

rgc2000, 2017-04-04:
Updated to head of master branch and added your files to /boot
This is the output after reboot : 
```
root@rpi2:~# vcdbg log msg |& grep SPARE
000837.548: SD: SPARE=400000, part=0
000843.155: PF: SPARE=400000, part=0
000877.251: PF: SPARE=400000, part=0
```

pelwell, 2017-04-04:
That's what I was afraid of - ho hum.

I've just been handed a Model B that supposedly exhibited this behaviour, but it has worked for me on the latest firmware, even under NOOBS.

Are you a NOOBS user?

rgc2000, 2017-04-04:
I'm using a raspbian wheezy on the 3 Model B.
Now, if this can help : 
- on the model B where boot is OK I only have 2 partitions on SD (partiton 0 /boot fat32, partition 1 / ext4)
- on models B where boot is not OK I have 3 partitions on SD (partiton 0 /boot fat32, partition 1 / ext4, partition 3 is an extended partition with 5 sub partitions using ext4)
I haven't tried to swap SD from one PI to another to see if boot is SD or Pi related. I can't do it now but I will this evening.

pelwell, 2017-04-04:
Good news - I have a failure here. I'll keep you posted.

dflogeras, 2017-04-04:
Fell into this as well, let me know if I can help test

pelwell, 2017-04-04:
The firmware update labelled "firmware: bootcode: Avoid PM_SPARE bit 23" should also have included new start.elf images, but didn't. [This zip file](https://drive.google.com/file/d/0B_P-i4u-SLBXWmVmUU12NzlmYk0/view?usp=sharing) contains those missing files, along with the "bad" bootcode.bin for convenience - can you give them a try?

rgc2000, 2017-04-04:
I've installed all files from your zip file in /boot
Pi is now booting OK. There is no debug message in vcdbg log.
Looks like you have found the cause. Thank you !

popcornmix, 2017-04-04:
Thanks @pelwell - that was my fault - I was thinking of that change as being bootcode only, but it did also affect start.elf.
I've pushed a rebuilt rpi-update firmware that should also work.

dflogeras, 2017-04-04:
I can confirm it works here with the .dat .elf and bootcode.bin from raspberrypi/firmware master

pelwell, 2017-04-04:
At last. Can we close this issue now?

rgc2000, 2017-04-04:
I've installed head master and all my Pi are booting ok now. Issue can be closed

closed 2017-04-04 by rgc2000

XECDesign, 2017-04-05:
@popcornmix Any chance of a /stable rebuild?

popcornmix, 2017-04-05:
Done.