Hexxeh/rpi-firmware #146: I was at 4.9.20, but after an apt-get upgrade, am at 4.4.50.

sbromberger, 2017-04-11:
Apologies if this is the wrong place to post this issue, but 10 minutes ago, I was at
`Linux pigpen 4.9.20-v7+ #985 SMP Mon Apr 3 10:30:44 BST 2017 armv7l GNU/Linux`

I did an `apt-get update` and `apt-get upgrade`:

```
seth@pigpen:~$ sudo apt-get update
[sudo] password for seth:
Get:1 http://archive.raspberrypi.org jessie InRelease [22.9 kB]
Get:2 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Get:3 http://mirrordirector.raspbian.org jessie/main armhf Packages [8981 kB]
Get:4 http://archive.raspberrypi.org jessie/main armhf Packages [145 kB]
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Get:5 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:6 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]
Get:7 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1356 B]
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9273 kB in 17s (527 kB/s)
Reading package lists... Done
seth@pigpen:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0
  raspberrypi-bootloader raspberrypi-kernel
6 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 66.6 MB of archives.
After this operation, 49.2 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```

Now I'm at
`Linux pigpen 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux`

This looks like a step back. What do I need to do?

thurask, 2017-04-11:
I'm not sure if this was intentional to roll back to 4.4, but deleting `/boot/.firmware_revision` and re-running `rpi-update` seems to do the trick.

sbromberger, 2017-04-11:
Thank you, @thurask - that fixed the problem! I still don't know why it happened, but I'm back at 4.9.20 now.

closed 2017-04-11 by sbromberger