Hexxeh/rpi-firmware #147: Notifications for release branch changes?

macmpi, 2017-05-05:
Hi,

Is there a way to be notified of critical branch family changes in the repo?

Recently `master` moved from 4.4x to 4.9.x, and it then followed last week with `stable` move from 4.4x to 4.9.x
It's all fine, but if one builds a project that is pulling firmware/kernel from one specific branch ('stable' for instance) of this repo, a new project build may all the sudden get into trouble because of unexpected significant change such as a (not-so)"minor" release leap.
(of course conservative approach is to pull by commit# rather than branch, but tracking by branch is also helpful).

A notification for such significant branch leaps in the repo (ideally with a short advance notice) would really be nice.
Thanks for consideration.

popcornmix, 2017-05-05:
No plans for this.
Keep an eye on the advanced users forum: https://www.raspberrypi.org/forums/viewforum.php?f=29
or follow [raspberry pi dev twitter](]https://twitter.com/rpf_dev_updates) if you want more information.

macmpi, 2017-05-06:
I'm afraid such specific information about this particular repo is very much buried in the forum & twitter: so it's unlikely this may provide an efficient way for tracking such events.
So, for a safe method,  developers have to fallback to regular manual commit#-specific integrations rather than tracking repo branches.

Ruffio, 2020-08-26:
Please close this issue.

Ruffio, 2020-12-27:
@popcornmix please close this request

closed 2020-12-27 by popcornmix