Hexxeh/rpi-firmware #148: Raspberry Kernel 4.9.25-v7+

RonaldoBi, 2017-05-16:
Good Morning:

Hello:
I used the Raspberry Pi 3 Model B, and i located in your site the kernal version 4.9.27-v7+.
I used the system Ubuntu 16.04.2.
In fact i installed and this version 4.9.27-v7+.
But i have the Lumenera câmera and this driver told me i need to install the version 4.9.25-v7+.
Could you help me to making  available this version for Raspberry?
Thank You.

Best Regards,

Ronaldo Bizarro

popcornmix, 2017-05-16:
I'm not really understanding.
Are you running rasbian jessie on the Pi and want to cross compile from Ubuntu?
Or are you running Ubuntu on your Pi?
What instructions are you following to use this lumenera driver?

RonaldoBi, 2017-05-16:
Good Morning:

Sorry, i used the Ubuntu Mate, because the driver for usb camera has a support for this system.
I used that Lumenera Camera Software Development Kit - Linux.
Now i found in your site the kernel version 4.9.25-v7.
I installed and don't have the error.
But when i use the command make to installl this driver has a error.
I don't understand.
The error is:

Coud you help me with this problem.
Thank you.

Best Regards,

Enviado do meu iPad

RonaldoBi, 2017-05-16:
> The error is:<image1.JPG>

Good Morning:

I forgot to explain where i founded the instruction for install this camera.
I created the directory Lumenera below my user.
I unzipped files. Have a readme with this instructions.
Thank you for your help.

Best Regards,

Enviado do meu iPad

popcornmix, 2017-05-17:
Ubuntu Mate uses their own kernel build so you should be asking them.
rpi-update is designed for raspbian users.

Ruffio, 2020-08-26:
@RonaldoBi please close this issue

closed 2020-08-27 by popcornmix