Hexxeh/rpi-firmware #149: firmware versions beyond 4.4.50 break i2c

Epposan, 2017-06-05:
I am having issues in communicating with an add-on board (ups_pico) on all my raspberry pi 2 model B v1.1 boards using i2c

* cat /proc/cpuinfo
hardware: bcm2709
revision: a01041

After a clean install of raspbian: **jessie-2017-04-10** (the latest image from the raspberrypi.org site)
* uname -a
4.4.50-v7+ #970
* enabled i2c
* installed ups_pico module
* i2cdetect -y 1
68 69 6a 6b	(expected output)
* i2cget -y 1 0x6b 0
0x5c			(firmware version)
* update /boot/config.txt with
dtoverlay=i2c-rtc,ds1307
* i2cdetect -y 1
UU 69 6a 6b	(rtc in use !)
  
  looks all fine!
  
After the initial apt-get update && apt-get upgrade
* uname -a
4.9.24-v7+ #993
* i2cdetect -y 1
68 69 6a 6b	(rtc not in use)
* i2cget -y 1 0x6b 0
0xff			(nothing read?)
  
  looking not so fine.
  
Using rpi-update to go back from **4.9.24** down to **4.9.11** (with several intermediate versions, though not all) did not bring i2c to live, however, after going back to **4.4.50**, everything was working as expected again.

Any ideas how to proceed from here?

pelwell, 2017-06-05:
This problem is solved by updating your Pico UPS firmware. See https://www.raspberrypi.org/forums/viewtopic.php?p=1121725#p1121725 .

Epposan, 2017-06-05:
Thanks for the link. I had already checked there. I have one HV3.0 ups_pico on which all is fine, however on my other pi's using HV1.0 and HV1.1 ups_pico's I *am* running the latest version (**0x5c**, dated 25-2-2016).

Nevertheless, it is good to know that the issue seems to be an issue with the ups_pico's and not with i2c in general.

In the message you refer to (dated feb 28, 2017) pimaster is talking about **a little delay** for the older hardware versions. At the moment of this writing we are already 4 months later and unfortunately, I have rather bad experiences with pimaster's predictions and priorities. Luckily the **safe shutdown** function of the older ups_pico's works fine, despite the failing i2c communication.

closed 2017-06-06 by Epposan