Hexxeh/rpi-firmware #150: Cannot boot after updating of RPi2 to 9e6a1a545ef33ac6cc3805845cb3ecac26514a41 (rainbow screen)

brozkeff, 2017-06-08:
After some time I ran rpi-update on a RPi2 that had fw already updated this way several times in the past without any issues. This time after reboot I get stuck at rainbow screen.

The working version of kernel before doing the update was according to dmesg output 4.1.15-v7+:

```
[    0.000000] Linux version 4.1.15-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #830 SMP Tue Dec 15 17:02:45 GMT 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: BCM2709
[    0.000000] cma: Reserved 8 MiB at 0x34800000
```

The messages shown during rpi-update:

```
pi@norbu:~$ sudo rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    101      0 --:--:--  0:00:01 --:--:--   101
100 53.9M  100 53.9M    0     0   178k      0  0:05:09  0:05:09 --:--:--  175k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.31+
 *** depmod 4.9.31-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 9e6a1a545ef33ac6cc3805845cb3ecac26514a41
 *** A reboot is needed to activate the new firmware
pi@norbu:~$ sudo reboot
Connection to norbu.local closed by remote host.
Connection to norbu.local closed.
```

SSH, ping failed after that, and when curious what happened and reconnected Rpi physically to HDMI screen just the rainbow image appeared.

Now I guess I have to downgrade fw to some earlier version, via 
```
sudo ROOT_PATH=/media/root BOOT_PATH=/media/boot rpi-update (hash-of-some-earlier-fw)
```

from my Linux box after mounting the microSD card.

Any idea why the latest fw failed? Are there some logs on the microSD card I could find to post them here so the reason why the RPi2 box gets stuck at rainbow boot image can be found?

popcornmix, 2017-06-08:
4.1 kernel was a long long time ago.
What distribution were you running? Raspbian Wheezy?
Can you post the contents of your config.txt and cmdline.txt

brozkeff, 2017-06-08:
It was originally wheezy but at some point I dist-upgraded it to jessie. Here I don't have any microSD card reader or microSD-to-SD adapter to be able to mount the card, I will do it at home later today.

popcornmix, 2017-06-08:
4.1.15 kernel is 18 months old, so it doesn't look like apt-get upgrade has been run (successfully) for a long time.
Still, config.txt options are the most likely - we'll see if there's anything suspicious there later.

popcornmix, 2017-06-08:
By the way, the hash for rpi-update to go back to that old kernel is:
```
sudo ROOT_PATH=/media/root BOOT_PATH=/media/boot rpi-update b8f4cb56a12ddeb047172ca9f99c7ea401dd5f50
```

brozkeff, 2017-06-08:
Thanks for the exact hash, will try this in the first place to get to a working box and then move on. 

There was not unattended-upgrades for all packages but only the security ones.

brozkeff, 2017-06-08:
cmdline.txt:

```
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
```

brozkeff, 2017-06-08:
config.txt:

```
# For more options and information see 
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#1360x768
#hdmi_group=2
#hdmi_mode=39
#1920x1080
hdmi_group=1
hdmi_mode=16

framebuffer_depth=32
framebuffer_ignore_alpha=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=1000
gpu_mem=160
dtparam=spi=off
dtparam=i2c_arm=off
#core_freq=500
#sdram_freq=500
#over_voltage=3
device_tree=
max_usb_current=1
```

pelwell, 2017-06-08:
Try removing `device_tree=` - current kernels require Device Tree.

brozkeff, 2017-06-08:
After downgrading the RPi2 boots again.

brozkeff, 2017-06-08:
Commenting device_tree= , updating via rpi-update again, and it works!

```
pi@norbu:~$ uname -a
Linux norbu 4.9.31-v7+ #1005 SMP Thu Jun 8 13:02:15 BST 2017 armv7l GNU/Linux
```

Thanks!

closed 2017-06-09 by popcornmix