Hexxeh/rpi-firmware #151: rpi-update f4cb84915874f32d88e2bfe793620d7afe25c755 sill stable for rpi2 unsing vdr and sundtek driver

flobee, 2017-06-11:
As a hint for you

rpi-update f4cb84915874f32d88e2bfe793620d7afe25c755 is still the only stable for rpi2 using vdr and sundtek driver. (but: stretch rpi repo)

All kernels above dosnt work and end up in mem errors. Where rpi is used only for vdr incl. sundtek hardware/drivers

Kind regards
flobee

popcornmix, 2017-06-12:
https://github.com/Hexxeh/rpi-firmware/commit/f4cb84915874f32d88e2bfe793620d7afe25c755 was the last commit on the BRANCH=next tree.
Can you confirm the same on the BRANCH=master tree?
5eb58b416697071f9187aa85cea23004ca0270f4 should be the same kernel revision.
Is that good? Does the next commit (8a68f57d6189074a476f7f2259b964bbd85d0143) break?

flobee, 2017-06-14:
Hi. Sry. I just use "rpi-update <hash>" and downgraded to a few week later where problems seems not exists. So i have not any branches... i will check right now... wait.

flobee, 2017-06-14:
Master branch i took, curently a recording is running, after it i will checkout your versions and will test it! thx for reply.

flobee, 2017-06-17:
report for 5eb58b416697071f9187aa85cea23004ca0270f4
running on: https://github.com/Hexxeh/rpi-firmware/commit/f4cb84915874f32d88e2bfe793620d7afe25c755

```
root@pi2 ~ # uname -a
Linux pi2 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
root@pi2 ~ # rpi-update 5eb58b416697071f9187aa85cea23004ca0270f4 # tests for github
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
Would you like to proceed? (y/N) y
*** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.13+
 *** depmod 4.9.13-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 5eb58b416697071f9187aa85cea23004ca0270f4
 *** A reboot is needed to activate the new firmware
rpi-update 5eb58b416697071f9187aa85cea23004ca0270f4  28,38s user 16,43s system 12% cpu 5:55,20 total
```

Status after the restart:
- dmesg: no problems
- sundek drivers started with:
```
    # For rpi
    # /opt/bin/mediaclient -P on -d /dev/dvb/adapter0/frontend0
    # /opt/bin/mediaclient -P on -d /dev/dvb/adapter1/frontend0
    # http://support.sundtek.com/index.php/topic,1573.0.html
    use_hwpidfilter=on
    # http://support.sundtek.com/index.php/topic,2215.0.html
    support_mmap=on
```
- Starting vdr: 
    Started, takes more than 60sec. to be ready. But ok, Is Up now.
- Testing recordings: 
    Two dongles, two different transponder for each. 
    Each dongle should record at the same time one channel using HD, one channel none HD channel 4 recordings at one time.
    Using (German Cabel TV): 
    Dongle 1: Das Erste HD (C), arte (C) 
    Dongle 2: ZDF HD (C), KiKA (C)

- Recording status: start
    duration 30min 19:00 - 19:30
    recordings startet, syslog silent
- Checks on fs: 
    recordings exists, 
    play check after 10 min using vlc: No issues, break detected
    I think at the moment i can stay with this kernel/ checkout

Summary after 20 min. It seems to fine.
 But wait for the next test which will be less documented but the same way

flobee, 2017-06-18:
report for 8a68f57
running on: 5eb58b4

Right, initially vdr gets up but cronjob later on trys to restart the vdr if it failts to start or record some timers. 

This version dont work at all.

The errors in logs / reports dont happen always. But more i can not find.

My custom log reports problems with hints i got from sundek to set in /etc/sundek.conf and this seems not to always to work this kernel:
```
Will run cmd: "/opt/bin/mediaclient -P on -d /dev/dvb/adapter0/frontend0"
2017-06-18 02:15:34 [root] [EMERG](0) Warning! Error code from shell execution detected, Set a high log level
2017-06-18 02:15:34 [root] [DEBUG](7) Cmd was: "/opt/bin/mediaclient -P on -d /dev/dvb/adapter0/frontend0"
2017-06-18 02:15:34 [root] [DEBUG](7) cmd code: 255
2017-06-18 02:15:34 [root] [DEBUG](7) cmd last line: device: /dev/dvb/adapter0/frontend0 doesn't support the extended media API
2017-06-18 02:15:34 [root] [DEBUG](7) cmd data[0]: Using device: /dev/dvb/adapter0/frontend0
2017-06-18 02:15:34 [root] [DEBUG](7) cmd data[1]: device: /dev/dvb/adapter0/frontend0 doesn't support the extended media API
2017-06-18 02:15:34 [root] [DEBUG](7) Will run cmd: "/opt/bin/mediaclient -P on -d /dev/dvb/adapter1/frontend0"
2017-06-18 02:15:34 [root] [DEBUG](7) Cmd was: "/opt/bin/mediaclient -P on -d /dev/dvb/adapter1/frontend0"
2017-06-18 02:15:34 [root] [DEBUG](7) cmd code: 255
2017-06-18 02:15:34 [root] [DEBUG](7) cmd last line: device: /dev/dvb/adapter1/frontend0 doesn't support the extended media API
2017-06-18 02:15:34 [root] [DEBUG](7) cmd data[0]: Using device: /dev/dvb/adapter1/frontend0
2017-06-18 02:15:34 [root] [DEBUG](7) cmd data[1]: device: /dev/dvb/adapter1/frontend0 doesn't support the extended media API
2017-06-18 02:15:34 [root] [DEBUG](7) Will run cmd: "systemctl start vdr.service"
2017-06-18 02:16:08 [root] [EMERG](0) Warning! Error code from shell execution detected, Set a high log level
2017-06-18 02:16:08 [root] [DEBUG](7) Cmd was: "systemctl start vdr.service"
2017-06-18 02:16:08 [root] [DEBUG](7) cmd code: 1
```

Syslog reorts too less but:
```
Jun 18 02:16:08 pi2 vdr[9778]: [9778] exiting, exit code 2
Jun 18 02:16:08 pi2 systemd[1]: vdr.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jun 18 02:16:08 pi2 systemd[1]: Failed to start Video Disk Recorder.
Jun 18 02:16:08 pi2 systemd[1]: vdr.service: Unit entered failed state.
Jun 18 02:16:08 pi2 systemd[1]: vdr.service: Failed with result 'exit-code'.
Jun 18 02:16:08 pi2 systemd[1]: Starting LSB: Starts vdradmin-am...
```

popcornmix, 2017-06-18:
Can you report output of:
```
uname -a
vcgencmd version
```
for both the working and not working updates.
I can't see anyhing in 8a68f57d6189074a476f7f2259b964bbd85d0143 that would cause a problem (the update didn't touch kernel and only affected composite progressive output modes), so I suspect the success and failure reports may be bogus.

flobee, 2017-06-19:
Hi 
I downgraded back to: 5eb58b4
uname -a:
`Linux pi2 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux`

vcgencmd version
```
Mar  1 2017 20:47:18 
Copyright (c) 2012 Broadcom
version 4abe87af4ec05aad4f6333001af4cded6390b64b (clean) (release)
```

hash 8a68f57 problem:
No real controll if vdr starts. Driver or Kernel. hmm!?
Ones yes (Ah sorry to tell: one recording i got and it was not really ok, but good enogh to watch, little breaks) but on the other hand: vdr did not start agian. 
Cronjob starts vdr but it dies :(
I lost recordings without good logs/messages except (5, 6 times that day): 
```
cmd last line: device: /dev/dvb/adapter0/frontend0 doesn't support the extended media API
2017-06-18 02:15:34 [root] [DEBUG](7) cmd data[0]: Using device: /dev/dvb/adapter0/frontend0
vdr[9778]: [9778] exiting, exit code 2
Jun 18 02:16:08 pi2 systemd[1]: vdr.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
```
Maybe a longer test will show more in syslog but this is already a no go for me. I have a feeling that mem error would come. But if vdr doesnt start, there is not reason to stay :/

kind regards 
flobee

flobee, 2017-06-26:
Just a hint. 
The bad: I downgraded to f4cb84915874f32d88e2bfe793620d7afe25c755 because i got to many bad recordings and no logs for a hint. :( 
This version seems to be fine for the last few days. If i find more i will report.

Ruffio, 2020-12-28:
@flobee  can this be closed?

flobee, 2021-01-05:
Hi! Yes, too long ago. i dont remember if i still running the older version
or a newer one. but my rpi is running since then ;-)

closed 2021-01-05 by popcornmix