Hexxeh/rpi-firmware #152: Are the stack clash fixes in the latest release?

Asmod4n, 2017-06-19:
Hi, are the fixes needed for https://blog.qualys.com/securitylabs/2017/06/19/the-stack-clash in the latest release?

popcornmix, 2017-06-20:
It sounds like the vulnerability affects a number of components.
It looks like some are fixed in debian (and so raspbian).
e.g. ``apt-get upgrade`` will get you a fixed version of sudo for example.
There is a libc fix which has reached debian but rasbian yet (but that is automated - we just need to wait for rebuilding to complete).

I think the kernel needs https://github.com/raspberrypi/linux/commit/1be7107fbe18eed3e319a6c3e83c78254b693acb which has reached the linux-4.12.y kernel, but not the linux-4.9.y tree yet. We will rebuild and push that once available.

popcornmix, 2017-06-20:
The libc fix has reached raspbian. I think we are just waiting on linux-4.9.y to be updated.

popcornmix, 2017-06-26:
Latest rpi-update kernel has the kernel side fixes e.g. https://github.com/raspberrypi/linux/commit/cfc0eb403816c5c4f9667d959de5e22789b5421e

so I think we are complete. Run 
```
sudo apt-get update
sudo apt-get upgrade
sudo rpi-update
```
to get all the latest fixes (although beware this gets testing firmware/kernel).
The testing firmware/kernel will reach apt-get soon.

popcornmix, 2017-07-01:
Kernel update is now in apt-get update.
Running apt-get upgrade will get you all stack clash related fixes.

popcornmix, 2017-07-01:
Actually - I think the kernel update hasn't reached apt. Should happen Monday.

Asmod4n, 2017-07-01:
might take some time till all mirrors have it, right?

Asmod4n, 2017-09-11:
Guess this is resolved :)

closed 2017-09-11 by Asmod4n