Hexxeh/rpi-firmware #153: Slower response time of USB-cardreader since 4.9.13

equino, 2017-07-20:
Ive attached a USB cardreader to my raspi. Up to version 4.9.11 the responsetime of the reader was about 115ms. Since 4.9.13 up to the newest one the time raised to 140-150ms.
When i downgrade to 4.9.11 everything is fine again....

popcornmix, 2017-07-20:
Can you confirm if this is due to gpu firmware or kernel?
e.g. Test with 4.9.11 kernel and gpu firmware from 4.9.13.

The gpu firmware consists of the files:
``start*.elf`` ``fixup*.dat`` ``bootcode.bin``
from ``/boot``.

equino, 2017-07-21:
Ok i upgraded to 4.9.13 again, but backed up the files you said. So the kernel and all other was 4.9.13 but the gpu firmware is from 4.9.11 (the last running one)
The result is the same, slow response of the USB reader...

In another board the speak about USB upstream changes in the 4.9 kernel?! I dont knwo excatly what they mean, but perhaps this is usefull for you. They also mentioned that in kernel 4.11 this "upstream issue" is fixed. Is there a plan from your side to bump kernel to 4.11?

Thanx in advance!

popcornmix, 2017-07-21:
Okay, so it is an upstream kernel issue.
Does the talk in the other board identify the upstream commit that fixes it in 4.11?
If it does we may be able to cherry-pick it.

Otherwise we provide source for kernel 4.11, 4.12 and 4.13, so you are free to build a kernel from these.
https://www.raspberrypi.org/documentation/linux/kernel/building.md

danyconnect, 2017-12-18:
i opened one ticket right here [https://github.com/raspberrypi/linux/issues/2295].
i not fixed in 4.14.5 yet.

equino, 2018-01-30:
According to the other ticket there are three commits concerning the ftdi.

427c3a9 USB: serial: ftdi_sio: fix modem-status error handling
c6dce26 USB: serial: ftdi_sio: fix extreme low-latency setting
a6bb1e1 USB: serial: ftdi_sio: fix line-status over-reporting

Perhaps we can do the "cherry-pick"? I can test it then...

Ruffio, 2020-12-27:
@equino is this issue still relevant? Else please close it, thanks :-)