Hexxeh/rpi-firmware #154: Pink line on raspberry pi screen left

coltrane312, 2017-08-11:
The line is about two pixels wide, by 600 long, across the left side

pelwell, 2017-08-11:
1. Is this a new issue? When did you first notice it? If you can narrow it down to a single firmware release that would help - see https://github.com/Hexxeh/rpi-firmware/commits/master for the list releases, and install a specific version using `sudo rpi-update <hash>` where `<hash>` is the string of hex digits shown on the right of each release.
2. On which type of screen do you see this?
3. Your description is a bit vague - a photo would be clearer; you can drag and drop one onto this issue.

Ruffio, 2020-12-27:
@coltrane312 is this issue still relevant? Else please close it, thanks :-)